wled/WLED #2051: Negative Segment Offset

Sturm2002, 2021-07-01:
Title says it all,
the possibility to enter a negative segment offset

V0.13.0-b0

huggy-d1, 2021-07-02:
Are you saying you want to enter a negative segment offset, or, you are submitting a bug report because entering a negative segment offset is allowed by the UI or API?

Sturm2002, 2021-07-02:
It would be nice to have the possibility to do so, because it's currently not possible

Aircoookie, 2021-07-03:
Hey! I've just added this in https://github.com/Aircoookie/WLED/commit/1bb7e36a6564805b424c76f2564146d21f978245 and it will be available in the next beta :)
UI and API now support entering a negative offset, although it will be converted to a positive one, taking the segment length into account (say you have a segment with 10 LEDs, an offset of `-2` will have the same effect as an offset of `8`)

closed 2021-07-03 by Aircoookie