wled/WLED #2053: API Call "&PL=~" not working

ElToberino, 2021-07-02:
**Describe the bug**
The API-Call "IP/win&PL=~" doesn't work at all, whereas the call of single presets with "IP/win&PL=5" or "IP/win&PL=6" works fine.

Is there an alternative way to realize this simple function (Preset++) with the JSON-API?

**WLED version**
 - Board: [ESP8266 - NodeMCU]
 - Version [beta version 0.13.0-b0]
 - Format [Binary]

Aircoookie, 2021-07-03:
Thank you!
It appears like I accidentally removed this feature together with preset cycle (which has been superseded by playlists)
Will restore it in the next beta release.
This is not yet supported in the JSON API, but planned: https://github.com/Aircoookie/WLED/issues/1933

ElToberino, 2021-07-03:
Thank you for your answer. So I'm looking forward to the next beta.

Btw: 
During the last years I've been testing a lot of different pixel controllers - for me there is no doubt that WLED is the most versatile and useful one. I would call it "the Swiss army knife among pixel controllers". Futhermore it is very well documented and supported. Thanks a lot for your work - it's really great!

whitey2861, 2021-08-10:
Hi. I am running 13.0. B2 and I am still having this issue with a macro of  pl=~.

The macro works and cycles, but only for the first 5 presets. Now that the option to change the preset cycle end point is gone it seems to think that the max is 5.  Does anyone have a solution to this? Or a different macro format  that will cycle through, let's say, 16 presets?

Thankyou

Aircoookie, 2021-08-18:
@whitey2861 hey! You can set the range with the P1 and P2 commands. Say you want to cycle through the 10 presets in the range 10 to 20, you can use `P1=10&P2=20&PL=~`

whitey2861, 2021-08-18:
Thankyou so much. That helps alot.

closed 2021-10-03 by blazoncek