wled/WLED #2056: Multiple pin LED outputs are not persistent across AC cycle

nolanhergert, 2021-07-04:
**Describe the bug**
I am able to get multiple LED strips (and respective segments) set up and working. I made sure to save to preset 16. When I AC cycle, the segment data comes back up, but only the first of the LED outputs on the LED Preferences page come back up. All other settings on that page persist fine.

**To Reproduce**
Go to LED Prefererences. Add a 2nd LED strip, hit save. Observe the 2nd LED strip is lit up. AC cycle. Observe 1st but not 2nd LED strip is lit up and only the first LED strip is shown on the LED Preferences page.

**Expected behavior**
Expect the 2nd LED strip to light up and be on the LED preferences page.

**WLED version**
 - Board: ESP32 w/ custom breakout
 - Version 13-b0 "Toki"
 - Format: Binary

**Additional context**
Everything is working except the persistence part! Thanks for your help!

nolanhergert, 2021-07-06:
Worked around by ensuring that Total LED count (on top of LED preferences page) is set to a value greater than the sum of the LED strips. 

A few options:
  * Update "Total LED count" value automatically to the sum of the LED strip lengths before submission
  * Make error saying the user needs to update this "Total LED count" value to be what is supported by their power supply/microcontroller. However, I would argue this should be handled by "maximum current" field and not LED count.

nolanhergert, 2021-07-06:
Created a pull request to facilitate discussion on solution #1 above: https://github.com/Aircoookie/WLED/pull/2065.

It is working for me when I add this line manually to html_settings.h, regenerate with npm, and re-upload to ESP32 dev board. No clue if it messes up other things though!

blazoncek, 2021-07-06:
This is something @Aircoookie explicitly does not want (ATM). There is a (legitimate) reason why it is currently possible to have discrepancy between actual amount of LEDs (sum of pixels on strips) and a total count in LED settings page.
It may be changed in the future when current reason has a workaround.

nolanhergert, 2021-07-06:
While I am ok with there being a discrepancy (I'm curious what the reason is, just for my understanding), part of the problem is that the code accepts a discrepancy initially (displays on the 2nd LED strip despite it being over Total LED count) but then rejects the discrepancy upon rebooting. It <understandably> looks like the checks are not consistently applied.

Aircoookie, 2021-07-06:
Hey! Thank you for the PR as well.
There are two reasons a discrepancy is possible:
- If you want to render a segment that is larger than the attached LEDs. This is useful for example if you want to have LEDs appear as one long strip, but driven by two separate (but UDP syncronized) controllers.
- If you want two strips displaying (partly) the same thing. You'd set both busses and the total to e.g. 100.

While I would like to retain this possibility, I do believe that having to set LED counts in two places can be unintuitive and error-prone. Luckily, there should be an easy solution. Currently, busses that exceed the set total LED count are just shortened to fit within it or are deleted entirely. However, I believe in case this happens and busses are configured to reach outside of the total, it is most likely that the user had intended for the busses to be this way and just forgot to update the total. Neither of the two use cases I described above require or support defining busses outside of the range specified by the total setting.

Therefore my proposal would be to extend the total LED count to the bus that ends at the highest index LED (start + length) in the `finalizeInit()` function :)
Therefore I'd propose

nolanhergert, 2021-07-07:
Sounds good to me! I tried messing around with the `_length `variable in `finalizeInit()` but it is still trimming the last LED strip like you said. I think I'll let you implement this one, if you don't mind. I'll reject the previous PR

Aircoookie, 2021-07-09:
Done! This was more involved than I thought, sorry for pointing you to `finalizeInit()`, which was in fact not the place this is handled 😅

commit 5da47636 (2021-07-09): Busses extend total configured LEDs if required (closes #2056 )

Fixed extra button pins defaulting to 0 on first boot
closed 2021-07-09 by Aircoookie