wled/WLED #2058: Indentation for json files in Editor

Sturm2002, 2021-07-04:
Hi, would it be possible to auto indent json files in the ESP Editor ? 

Instead of having the hole file contents in one row, having them in multiple rows and with proper indentation

**Example**

Before (Everything in one row):
{"glossary": {"title": "example glossary",GlossDiv": {"title": "S","GlossList": {"GlossEntry": {"ID": "SGML","SortAs": SGML",GlossTerm": "Standard Generalized Markup Language","Acronym": "SGML","Abbrev": "ISO 8879:1986","GlossDef": {"para": "A meta-markup language, used to create markup languages such as DocBook.","GlossSeeAlso": ["GML", "XML"]},GlossSee": "markup"}}}}}

After (Indented):
{
    "glossary": {
        "title": "example glossary",
		"GlossDiv": {
            "title": "S",
			"GlossList": {
                "GlossEntry": {
                    "ID": "SGML",
					"SortAs": "SGML",
					"GlossTerm": "Standard Generalized Markup Language",
					"Acronym": "SGML",
					"Abbrev": "ISO 8879:1986",
					"GlossDef": {
                        "para": "A meta-markup language, used to create markup languages such as DocBook.",
						"GlossSeeAlso": ["GML", "XML"]
                    },
					"GlossSee": "markup"
                }
            }
        }
    }
}

blazoncek, 2021-07-06:
One character at a time can consume a lot of flash. ;)
This (editor) is a part of AsyncWebServer library, so maybe direct the question there...

Aircoookie, 2021-07-06:
That is so true :) Internally we'll always want to store the data in a single line.

The /edit page seems to use the ACE editor, and it might be possible for it to auto prettify the JSON: https://stackoverflow.com/questions/31767051/how-do-i-use-beautify-in-ace-editor/31767057

I do in fact have my own fork of AsyncWebServer, which is used in WLED :) So we could definitely change the edit page ourselves.
https://github.com/Aircoookie/ESPAsyncWebServer

Otherwise, I can highly recommend https://jsonlint.com/ 
It not only indents the JSON, but also makes sure it is valid :)

Sturm2002, 2021-07-06:
Alright, i will look into that, thank you both ^^

closed 2021-07-06 by Sturm2002