wled/WLED #2059: build error: "fatal error: ESPAsyncUDP.h: No such file or directory"

fl380, 2021-07-05:
**Describe the bug**
"fatal error: ESPAsyncUDP.h: No such file or directory" when compiling

**To Reproduce**
./build custom

[env:custom_d1_mini_usermod_dallas_temperature_C]
extends = env:d1_mini
build_flags = ${common.build_flags_esp8266} -D USERMOD_DALLASTEMPERATURE -D USERMOD_DALLASTEMPERATURE_CELSIUS
lib_deps = ${env.lib_deps}
    milesburton/DallasTemperature@^3.9.1
    OneWire@~2.3.5

**Expected behavior**
successful compilation

**WLED version**
 - Board: d1_mini
 - Version 
 - Format Discord Bot

**Additional context**
```
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 (2.6.2) > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20704.7 (2.7.4) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Converting wled00.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ strict
Library Manager: Installing fastled/FastLED @ 3.4.0
Library Manager: FastLED @ 3.4.0 has been installed!
Library Manager: Installing IRremoteESP8266 @ 2.7.18
Library Manager: IRremoteESP8266 @ 2.7.18 has been installed!
Library Manager: Installing git+https://github.com/lorol/LITTLEFS.git
git version 2.20.1
Cloning into '/root/.platformio/.cache/tmp/pkg-installing-f68dq04h'...
Library Manager: LittleFS_esp32 @ 1.0.6+sha.f3bce17 has been installed!
Library Manager: Installing git+https://github.com/Aircoookie/ESPAsyncWebServer.git @ ~2.0.2
git version 2.20.1
Cloning into '/root/.platformio/.cache/tmp/pkg-installing-mx7rcyjm'...
Library Manager: ESP Async WebServer @ 2.0.2+sha.d1a267c has been installed!
Library Manager: Installing dependencies...
Library Manager: Installing ESPAsyncTCP
Library Manager: ESPAsyncTCP @ 1.2.2 has been installed!
Library Manager: Installing AsyncTCP
Library Manager: AsyncTCP @ 1.1.1 has been installed!
Library Manager: Installing OneWire @ ~2.3.5
Library Manager: Warning! More than one package has been found by OneWire @ ~2.3.5 requirements:
 - paulstoffregen/OneWire @ 2.3.5
 - ivankravets/OneWire @ 0.0.0+sha.7bd102cb73e0
 - mbed-alpov/OneWire @ 0.0.0+sha.445fe6e6bd68
 - mbed-hudakz/OneWire @ 0.0.0+sha.7f7759008807
 - mbed-fblanc/OneWire @ 0.0.0+sha.8c4e1841eb30
 - mbed-sroy/OneWire @ 0.0.0+sha.7c6dd6bc20e4
 - mbed-simonbarker/Onewire @ 0.0.0+sha.d961f715d82b
 - mbed-waf/OneWire @ 0.0.0+sha.79a1b7700cf2
 - mbed-jack-zen/OneWire @ 0.0.0+sha.a47e8ec71a4e
Library Manager: Please specify detailed REQUIREMENTS using package owner and version (showed above) to avoid name conflicts
Library Manager: OneWire @ 2.3.5 has been installed!
Library Manager: Installing milesburton/DallasTemperature @ ^3.9.0
Library Manager: DallasTemperature @ 3.9.1 has been installed!
Library Manager: Installing dependencies...
Library Manager: OneWire @ 2.3.5 is already installed
Library Manager: DallasTemperature @ 3.9.1 is already installed
Found 35 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <FastLED> 3.4.0
|   |-- <SPI> 1.0
|   |-- <EspSoftwareSerial> 6.8.5
|-- <IRremoteESP8266> 2.7.18
|-- <ESP Async WebServer> 2.0.2+sha.d1a267c
|   |-- <ESPAsyncTCP> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <LittleFS(esp8266)> 0.1.0
|-- <OneWire> 2.3.5
|-- <DallasTemperature> 3.9.1
|   |-- <OneWire> 2.3.5
|-- <ESPAsyncTCP> 1.2.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266WebServer> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <Wire> 1.0
|-- <ArduinoOTA> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS> 1.2
|   |   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
|-- <EEPROM> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <LittleFS(esp8266)> 0.1.0
Building in release mode
Compiling .pio/build/custom_d1_mini_usermod_dallas_temperature_C/libc15/FastLED/FastLED.cpp.o
Compiling .pio/build/custom_d1_mini_usermod_dallas_temperature_C/libc15/FastLED/bitswap.cpp.o
Compiling .pio/build/custom_d1_mini_usermod_dallas_temperature_C/libc15/FastLED/colorpalettes.cpp.o
Compiling .pio/build/custom_d1_mini_usermod_dallas_temperature_C/libc15/FastLED/colorutils.cpp.o
Compiling .pio/build/custom_d1_mini_usermod_dallas_temperature_C/libc15/FastLED/hsv2rgb.cpp.o
Compiling .pio/build/custom_d1_mini_usermod_dallas_temperature_C/libc15/FastLED/lib8tion.cpp.o
Compiling .pio/build/custom_d1_mini_usermod_dallas_temperature_C/libc15/FastLED/noise.cpp.o
Compiling .pio/build/custom_d1_mini_usermod_dallas_temperature_C/libc15/FastLED/platforms.cpp.o
Compiling .pio/build/custom_d1_mini_usermod_dallas_temperature_C/libc15/FastLED/platforms/esp/32/clockless_rmt_esp32.cpp.o
Compiling .pio/build/custom_d1_mini_usermod_dallas_temperature_C/libc15/FastLED/power_mgt.cpp.o
Compiling .pio/build/custom_d1_mini_usermod_dallas_temperature_C/libc15/FastLED/wiring.cpp.o
Compiling .pio/build/custom_d1_mini_usermod_dallas_temperature_C/src/FX.cpp.o
In file included from wled00/wled.h:80:0,
                 from wled00/FX.h:27,
                 from wled00/FX.cpp:27:
wled00/my_config.h:17:2: warning: #warning **** my_config.h: Settings from this file are honored **** [-Wcpp]
 #warning **** my_config.h: Settings from this file are honored ****
  ^
In file included from wled00/wled.h:110:0,
                 from wled00/FX.h:27,
                 from wled00/FX.cpp:27:
wled00/src/dependencies/e131/ESPAsyncE131.h:33:25: fatal error: ESPAsyncUDP.h: No such file or directory

*********************************************************************
* Looking for ESPAsyncUDP.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESPAsyncUDP.h"
* Web  > https://platformio.org/lib/search?query=header:ESPAsyncUDP.h
*
*********************************************************************

 #include <ESPAsyncUDP.h>
                         ^
compilation terminated.
*** [.pio/build/custom_d1_mini_usermod_dallas_temperature_C/src/FX.cpp.o] Error 1`

srg74, 2021-07-05:
This should work for you
`[env:custom_d1_mini_usermod_dallas_temperature_C]
board = d1_mini
platform = ${common.platform_wled_default}
platform_packages = ${common.platform_packages}
board_build.ldscript = ${common.ldscript_4m1m}
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags_esp8266}
  -D USERMOD_DALLASTEMPERATURE
  -D USERMOD_DALLASTEMPERATURE_CELSIUS
  -UWLED_USE_MY_CONFIG
monitor_filters = esp8266_exception_decoder
lib_deps = ${esp8266.lib_deps}
  OneWire@~2.3.5
  milesburton/DallasTemperature@^3.9.0`

fl380, 2021-07-05:
> This should work for you
> `[env:custom_d1_mini_usermod_dallas_temperature_C] board = d1_mini platform = ${common.platform_wled_default} platform_packages = ${common.platform_packages} [...]

thanks, unfortunately also fails but there's no log this time, "404: Not Found" when I click on the link to the log. My config worked a few days ago so suspect that it's a temporary problem

srg74, 2021-07-05:
> thanks, unfortunately also fails but there's no log this time, "404: Not Found" when I click on the link to the log. My config worked a few days ago so suspect that it's a temporary problem

Clean your .pio and ./buildcache and everything should be fine.

fl380, 2021-07-05:
> Clean your .pio and ./buildcache and everything should be fine.

I'm using the Discord bot so I don't think that it is possible

srg74, 2021-07-05:
Go to Discord. You build is ready.

closed 2021-07-05 by fl380