wled/WLED #2064: Ethernet support does not allocate pins

henrygab, 2021-07-06:
**Describe the bug**

Ethernet configuration does not check if the pins are already allocated, and does not allocate pins to prevent other use.

**To Reproduce**

Use ethernet-enabled binary.
Send configuration that sets relay to pin 12.
Click power button in UI

**Expected behavior**

LEDs turn off, but not the board/ethernet.

**Actual behavior**

See [thread](https://github.com/Aircoookie/WLED/issues/1027#issuecomment-723481641) ... powers off because ethernet [did not reserve](https://github.com/Aircoookie/WLED/blob/8c44147a4596112a7f5d5720ae809949399e9d44/wled00/wled.cpp#L438-L448) the pins it needs.

**WLED version**
 - Board: ESP32-POE
 - Version: 0.12.0
 - Format: binary

**Additional context** 

Also would occur on other ethernet boards. Fix may be easy.  I will see about adding a PR, but will need help testing it.

Thank you for your help!

blazoncek, 2021-07-06:
@Aircoookie shall I make a PR for this? I have it already in my fork.

Aircoookie, 2021-07-06:
@henrygab thank you for #2066 and @blazoncek thanks for the review :) Feel free to make a 2nd PR if it is a different approach. But I think #2066 is fine with the points in your comments addressed

blazoncek, 2021-07-07:
My fork (prior to the catch from @henrygab) only dealt with Settings page reservations (so pins couldn't be allocated at 1st place).
I've updated it now to include reservations at a proper place now (cfg.cpp) for actual pinManager allocations. It also handles Usermod settings by adding pins to cfg.json.
In its current form it does not handle changes of Ethernet type properly & requires reboot after change.

henrygab, 2021-07-07:
OK, I've marked my PR as ready for review, and also fixed another ethernet-related bug (cleanup when `ETH.begin()` returns false).  I have not tested the changes on hardware (yet), due to time constraints.

PR #2066 (2021-07-09): Bug 2064, 2063 - PinManager usage
### CURRENTLY NOT BUILT OR TESTED

However, ready for review and discussion of proper way to fix.

Fixes #2063 - Bus Manager classes `free()`ing pins it shouldn't

The fix is to only store the pin number in the private class variable **_after_** allocation succeeds.

Fixes #2064 - Ethernet support should allocate pins using `PinManager` class

The fix is to attempt to allocate the pins, and track the ones that were successfully allocated, so can safely roll-back if all pins were not allocated.  Only when all are successfully allocated, then initialize ethernet [... cut]
closed 2021-07-09 by Aircoookie