wled/WLED #2075: Multiple WLED devices

CLMNTDSGNR, 2021-07-12:
Hello,

First of all I would like to thank Aircookie who did a great job on WLED!

However I have a little problem: /
Yesterday I installed a new device in addition to the first one I installed a few months ago.

I have two devices, Kitchen and Bedroom.
Everything is well discovered by Alexa, no worries on this side. The problem is that when I ask Alexa to turn off "Kitchen" it turns off my 2 devices (Kitchen and bedroom). Same when I ask Alexa to turn off "Bedroom". Bedroom and kitchen go off.

I searched, tweaked a few things but nothing changes: /

Thanks for your help

closed 2021-07-12 by CLMNTDSGNR

numanair, 2022-01-18:
Were you able to resolve this? I am having the same problem.

numanair, 2022-01-18:
A few hours later and it is back to not working correctly.