wled/WLED #2077: Error compiling WLED for usermod staircase

sagaardu, 2021-07-14:
Hello,

I am trying to compile WLED to use the staircase usermod but it gives me an error compiling.

I already have the indicated libraries installed and I use core 2.7.4

The IDE highlights the following line for me

```
exit status 1
there are no arguments to 'NeoTm1814Settings' that depend on a template parameter, so a declaration of 'NeoTm1814Settings' must be available [-fpermissive]
```

huggy-d1, 2021-07-15:
Are you using VS Code + PlatformIO IDE, or Arduino IDE, or something else?  Preferred and dead simplest approach is VS Code + PlatformIO IDE.  2 build attempts is all it takes for all libraries to resolve dependencies.  The first build downloads all the libraries.  The 2nd build actually builds the firmware.

blazoncek, 2021-10-06:
Get help for compiling on Discord.
Closing.