wled/WLED #2079: PIR night day; time selected button preset

Rushmed, 2021-07-16:
Currently I am using precompiled 0.13.0-b2 on a Wemos mini with an AM312 PIR attached.
Motion detection works, LEDs are automatically switched on an off one minute later using nightlight in the triggered preset.
Thats great so far. But it`s doing this all day long. So I am very intrersted in the ability to switch on/off motion detection according to daytime or even better to sunrise and sunset.
This functionaltiy is not included in the usermods an I dont want to compile my own version.
Is there a way using presets? Tried saving different presets each sunrise and sunset to the preset that is triggered by PIR but the off preset was not working properly.

So my suggestion is to implement a column in the "Time-controlled presets" table to set the button (PIR) triggered preset according to time.
[image]

Of courste thats jus my idea. If theres another present way to enable/disable motion detection according to time or ambient light I would be very thankful.

Rushmed, 2021-07-20:
Ok, thanks. I compiled WLED with PIR_sensor_switch usermod.
Now I am a litte confused. 

First it always switches lights on when motion is detected, even if I uncheck "PIRenabled:".
But it dosent turn off. 
If I check "PIRenabled:" it works as expected.
How to switch motion detection off?

Second. It makes no differece if "nighttime-only:" is checked or not. Time is set up correctly. I even tried setting current time to a time that is definitively daytime without success. it always switches at motion detection.

I cleared storage before flashing.

What am I doing wrong?

blazoncek, 2021-07-20:
If it is not enabled it _will not_ interfere with WLED. I have tested it that much. 😄 
Check your GPIO settings and wiring. Usermod has been tested with SR501 and SR602 PIR sensors and works as expected.
With SR501 you can adjust the sensitivity and duration of "on" state with trimmers. The amount of PIRoffSec will add to the adjusted sensor time.
For nighttime-only you have to set NTP time and timezone and geo location (latitude & longitude) so that sunrise and sunset can be calculated.

Rushmed, 2021-07-20:
I am still testing. Had not set the geo location. Now with location set nighttime is working as expected. Wiring is okay, I went back to default pin 13 with same behavoir. Still every motion enables light without swichtching off after timeout.
Again to be clear:
   "PIRenabled:" checked: everything ok, switching and an off as expected.
   "PIRenabled:" unchecked: LEDs switch on when detecting motion without switching off.

I prefer using the AM312 because of its small footprint  and very reliable experience in other projects. I have some SR501 laying around and can do some comparison tests maybe tomorrow. 
SR501 has extremely often false positives in the other projects i did which i could only fix wit this [wiring workaround](http://techgurka.blogspot.com/2013/05/cheap-pyroelectric-infrared-pir-motion.html) 
Tested the SR501 so far on WLED with default wiring and had the same massive false positive issue.

By the way thanks for writing, developing and supporting your usermod and of course thanks to @Aircoookie for this superior project!

blazoncek, 2021-07-21:
What are your on/off presets? If both are 0, the PIR is just switching power on/off, otherwise it triggers that particular preset.
If you disable usermod and still see WLED swiching on (and not off) it may be unrelated to usermod. Disabled usermod does not do anything (not even initialise the sensor), but you can also set pin to -1 which will disconnect it on a HW level.
If you use MQTT you can monitor `/wled/#` topic for `/motion` messages which will fire when usermod reacts to motion.

Rushmed, 2021-07-21:
My presets are &T=0 for off and &T=1 for on. If I switch them in usermod setting the behavoir is inverted but again motion is triggered but LEDs do not go back to initial state when "PIRenabled:" unchecked.  Same with presets set to 0. 
MQTT motion message is always sent independend from "PIRenabled:".

blazoncek, 2021-07-21:
If you uncheck PIRenabled, the usermod has NO FUNCTION.

Rushmed, 2021-07-22:
Exactly this is NOT working for me, maybe something went wrong while compilating.

blazoncek, 2021-07-22:
I really have no idea how this piece of code could not work as intended:
```
  void loop()
  {
    // only check sensors 4x/s
    if (!enabled || millis() - lastLoop < 250 || strip.isUpdating()) return;
```

Rushmed, 2021-07-22:
Where to finde the code, don`t see it in any file @ https://github.com/Aircoookie/WLED/tree/master/usermods/PIR_sensor_switch

blazoncek, 2021-07-22:
Looks like not all changes made it to master.
Oops. 😄

closed 2021-07-22 by Aircoookie

Rushmed, 2021-07-22:
I glad that we found the issue. Thanks genteleman's.

PR #2090 (2021-07-22): Fix for not honouring enabled state for PIR usermod.
This should fix #2079