wled/WLED #2082: password ?

Flodr, 2021-07-18:
I would like to know, and if possible, how to put a password on the web interface and the requests api.

Thank you.

Sturm2002, 2021-07-20:
As far as i know, it is currently not possible, correct me please if i'm wrong.
try to open a Feature Request maybe ?

Flodr, 2021-07-20:
Alright thank you, but I'll wait a little longer to see if anyone else knows how to do it, else i will do a feature request.

pbolduc, 2021-07-23:
The library that does the web server does support authentication. We would need to be able to save the username and password entered and validate on each request. If you forgot the password, you would effectively locked out and would probably have to erase the flash (and all presets, etc) to gain access again. Reference: https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WebServer#authentication

Flodr, 2021-07-23:
Thanks ! I close this issue now.

closed 2021-07-23 by Flodr

pbolduc, 2021-07-23:
The other non-http services like UDP sync or E131 can't have passwords due to the nature of those interfaces. 

Also we would need to evaluate any external integrations like home assistant. They would need to support supplying a password.

What is your use case? Prevent kids/roommate from messing with the settings on your home network?  I could see a need to secure the endpoint that allows changes. I wouldn't want my kids turning on led lights in my bedroom when I am sleeping.

There are more advanced/complex ways like using a reverse proxy like traefik, but I understand that can be too much for a novice.

I think if the scope is initially constrained, this something I could look at working on in the coming weeks.

jn3va, 2021-12-20:
This would be great feature to add, and might be essential these days.   A friend is using this and the wifi at his apartment is open and shared between several units so there is the potential for people to mess with his light.  You could make this optional like you do with the buttons -1 means not configured and a password means configured.  Thanks!