wled/WLED #2083: Control multiple LEDs as a single LED (such as in Effects)

lossofworth, 2021-07-19:
Hello,

My project is hex shaped, LED lit panels. I feel like it's a pretty common project. I have 6 LEDs in 10 panels making 60 individually addressable LEDS. I would like each of these panels to act as one LED when using effects such as Rainbow Chase. Is this possible?

Aircoookie, 2021-07-19:
Yes, of course! Just set the Grouping setting in Segment 0 to 6 :)

lossofworth, 2021-07-20:
> Yes, of course! Just set the Grouping setting in Segment 0 to 6 :)

So I've setup 10 segments each in their own group. Then I selected the effect Chase. With this the individual LEDS are changing instead of each group changing at the same time. Advice?

blazoncek, 2021-07-20:
He said Grouping set to 6 in Segment 0. Not 10 segments.
You only need 1 segment.

Sturm2002, 2021-07-20:
As they already said, try to not create multiple segements, but rather one (Segment 0) and just set the spacing in the segment options to, in your case 6, aslong as you don't have spare LEDs in between them.
Try reading the Segments Part in the Wiki : https://github.com/Aircoookie/WLED/wiki/Segments

lossofworth, 2021-07-20:
🤦 Thank you for that. I definitely didn't RTFM on what exactly grouping did. It now works!

closed 2021-07-20 by Aircoookie