wled/WLED #2086: First LEDs turn on by themselves

robyevolution, 2021-07-20:
Hi everyone I have many fully functional wled strips, but I am having problems with the last strip, a ws2812b with 376 leds, controlled by a wemos d1 mini with version 0.12 (I also tried 0.13 beta).
When I turn off the strip, then the first 10 LEDs light up by themselves, how can I solve?
I also have another problem, when I set the white brightness to maximum, the last 70-80 leds turn green, if I turn down the brightness they turn white back, what could it be? I set the LEDs to 35mv efficiency, and the power supply is a 16A meanwell.

blazoncek, 2021-07-21:
First issue is most likely faulty LEDs.
Second issue is related to inadequate power. Use power injection on your strip.

poelzi, 2021-07-21:
you can try #1968 , it fixed this problem for me

robyevolution, 2021-07-21:
how do i edit the file?

xawwnee, 2021-07-26:
> Hi everyone I have many fully functional wled strips, but I am having problems with the last strip, a ws2812b with 376 leds, controlled by a wemos d1 mini with version 0.12 (I also tried 0.13 beta). [...]

WS2812B: 376 x 0.06A = 22.56 Amps (Min. required)
Use 5V 25Amps supply.

blazoncek, 2021-10-06:
Since this is not a bug but a HW problem (possible noise on data line, power issue, etc) I am closing.
Feel free to reopen if you think this is indeed SW issue.

closed 2021-10-06 by blazoncek