wled/WLED #2087: Button gedrückt halten um einen bestimmten Effekt laufen zu lassen.

Ronny-1979, 2021-07-21:
Hallo,

mich schreibe mal in deutsch, englisch ist nicht so das es alle verstehen würden.  

Ist es möglich wenn man einen Button gedrückt hält das da ein bestimmter Effekt abgespielt wird?  Nach dem loslassen des Buttons müsste es dann wieder auf die Playlist zurückfallen. 

Gruß Ronny

Sturm2002, 2021-07-29:
In English,
***
Hi,
i'm gonna write on german, due to my bad english.

would it be possible to play a effect only while holding a button, if i release it, it's gonna switch back to the Playlist

best regards
Ronny
***

Sturm2002, 2021-07-29:
@Ronny-1979 probier mal die seite deepl.com aus, klatsch da deinen deutschen text rein, und poste hier die englische übersetzung, das ist ein "englischen projekt" auf deutsch zu schreiben bringt hier nix ;D
***
try deepl.com out, write your german text there, and use the converted text to post it here, it's not useful to write on german, because this is a "english project"

Ronny-1979, 2021-07-29:
Hat sich erledigt, man müsste bloß mal richtig lesen. Über Switch funktioniert es. 

That's taken care of, you just have to read it properly. It works via Switch.

closed 2021-07-29 by Ronny-1979