wled/WLED #2088: Playlist pseudo element

poelzi, 2021-07-21:
**Is your feature request related to a problem? Please describe.**
I love the new playlist feature, very usefull. Unfortunatelly, if you have lots of presets, it became much harder to handle.

**Describe the solution you'd like**
I suggest a new pseudo preset "--- All Presets ---" that expands in memory to a list of all presets on the controller with the same transition settings as this entry. This way, the old behaviour could be simulated with one entry, shuffle would also be easily possible.

If presets could have tags, the pseudo element could represet all presets that have a certain tag.

huggy-d1, 2021-07-29:
Presets can be given shortcut names in the favorites section / tab.  That may not help in your "all presets" request with "random, play all presets, transition timing = xx.xx sec", but it might help for other uses.

blazoncek, 2023-08-05:
The problem with presets is that they are never stored in RAM (list) so unless you read and parse the presets.json file (tedious and time consuming) it is not possible to achieve such feature.