wled/WLED #2089: End Playlist should allow playlist

poelzi, 2021-07-21:
**Is your feature request related to a problem? Please describe.**
After the playlist has reached it's end, only a preset can be selected.

**Describe the solution you'd like**
It would be great, if also another playlist could be select, which would load the playist like a normal load would. This way, complex scenarios could be build.

blazoncek, 2021-07-22:
Playlists are regular JSON API presets as any other preset you make using GUI.
Nothing is preventing you to manually assign a _new_ playlist preset ID to end preset in _old_ playlist. Uncheck "Show playlist editor" and edit JSON.
In this way you can chain multiple playlists.

blazoncek, 2021-10-06:
Added and will be available in next beta release.