wled/WLED #2097: Physical DMX in

darranwil, 2021-07-24:
**Is your feature request related to a problem? Please describe.**
The ability to add a max-485 and jack to feed physical DMX lines into WLED.

**Describe the solution you'd like**
The ability to add a max-485 and jack to feed physical DMX lines into WLED to control LED strips.

**Describe alternatives you've considered**
Haven't found anything 

**Additional context**
Want to upgrade some 12v non addressable RGBW bars, DMX controlled, at the venue I work at with addressable LED tape. Since these are part of the "House Plot" we let touring Lighting Designers plug into (Single DMX universe) I would like to use WLED with physical DMX line in/out (out just bridged to in). Ideally with the already developed option to use entire strip as single fixture or addressable, or each wled defined segment as a DMX fixture. 

Thank you for your ideas for making WLED better!

Barabba11, 2021-07-27:
this is great idea! I fully support it, I guess there are already projects and schemes for it, it may help a lot to introduce it here easly

Barabba11, 2021-07-27:
WHen I built up a 8266 bridge Artnet to DMX I've tired different projects, but only this gave me good results:
https://github.com/bombcheck/ESP8266_ArtNet-LED-DMX-Node
I think it contain already everything needed to manage the hardware communication

Barabba11, 2021-07-28:
seems that there aren't hardware conflicts between the 2 projects, WLED uses GPIO2, D4 pin to output the led while bombcheck uses GPIO1, RX pin, to receive from DMX line. perhaps merge the codes isn't a big problem.
All I've found for receive on DMX on his code is:
```
void dmxIn(uint16_t num) {
  // Double buffer switch
  byte* tmp = dataIn;
  dataIn = dmxA.getChans();
  dmxA.setBuffer(tmp);
  
  newDmxIn = true;
}
```
and all seems already into library espDMX_RDM.h

Aircoookie, 2023-01-02:
Closing in favor of #2824

closed 2023-01-02 by Aircoookie