wled/WLED #2099: Saving Segment Settings?

xawwnee, 2021-07-26:
Segment settings gets reset every time on esp on/off, is it possible to store segment settings?

blazoncek, 2021-07-26:
Save a preset.
And look up Wiki.

xawwnee, 2021-07-27:
> Save a preset.
> And look up Wiki.

Saving preset is working, I am asking for last used config for segments and other customizations.

blazoncek, 2021-07-27:
Use Auto Save usermod.
Has to be custom compiled in.

xawwnee, 2021-07-27:
> Use Auto Save usermod.
> Has to be custom compiled in.

Thanks for confirming, can you check your mail or twitter?

blazoncek, 2021-07-27:
Why? I do not reply privately.

xawwnee, 2021-07-27:
> Why? I do not reply privately.

I have other few questions that's why, ok if you're comfortable with private reply. I am asking you here.

A single 5V 2.1A is being injected to ESP8266, MAX4466 and  WS2812B (71 Nos), whether the given supply is safe for each module.
Issue: Strip flickers (first 1 or 2 led) before it turns on to orange.

No resistors / capacitors being used.

I am planning to change supply to 5V 5A (as my strip requires 4.26A)

Using this diag. https://i.imgur.com/yOa9FT6.png

blazoncek, 2021-07-27:
Please join WLED Discord for support questions.

closed 2021-08-18 by Aircoookie