wled/WLED #2102: Sync infrastructure / Instance sync disable

Sturm2002, 2021-07-29:
**Describe the solution you'd like**
having the possibility to select which instances to sync to which one, for example instance 0 is the "master" and 1 and 2 should sync to that one, but in your other room you have another instance 3 which is also a "master" and 4 and 5 should sync to instance 3

**Describe alternatives you've considered**
another sync option (the button on the top left) wether exactly this instance should sync at all, so you could sync for example 2 out of 3 instances and the third is a "standalone" instance.

**Additional context**
[image]

Thank you for maintaining this wonderful project <3

huggy-d1, 2021-07-29:
In instance 0, 1, 2, leave the UDP sync port # at the default value.
In instance 3, 4, 5, change all UPD sync port #'s, say to one port # higher.
Then they are independent of each other.
It's all about the sync port # as to which devices talk to one another.

Sturm2002, 2021-07-29:
Oh, i didn't know that. thank you very much ^^

closed 2021-07-29 by Sturm2002