wled/WLED #2103: Inverter problems in Camper

Kaistee93, 2021-07-29:
Hello there :)
I build myself a camper on top of a trailer and wanted to illuminate it with WLED because i can set different colors to Segments and the awesome effects. My problem - the whole stripe is flickering in various colors (when set to solid orange). The problem only occurs if the Inverter is turned on or charging the batteries. So i watched the videos on how to solve flickering stripe. The cables to the Stripe are 1,5mm² and it's not only one wire going up to the segments but every two segments have an input cable. Voltage drops by maybe 0,5V at the end of the Stripe First the ESP was down in the Powerbox, then i moved it directly infront of the first segment with a thic data Wire going out and all grounds connected. I tried different modules (WeMos D1 mini, NodeMCU V3) levelshifters and increased the 5V and 12V to 5.2V and 12.9V. The segments datalines are connected by an old CAT7 network cable.

My Setup is as follows:
  4in1 Inverter Ective SSI10 24V
  2*2 SolarPanels (18V, 100W in Series x2) hooked up to the inverter
  2* 110AH 12V Solarbattery in Series hooked up to the inverter
  Buck converter 30Amp 24V to 12V (for the stripe and a small water Pump)
  Buck converter 5Amp 12V to 5V (for the ESP-Board)
  NodeMCU V3 with WLED beta version 0.13.0-b2
  RelaisBoard to switch +12V and GND on and off (Because sometimes the leds suddenly turned on while esp was off)
  IR receiver and two Switches wired parallel to turn the system on/off
  10meters WS2815 RGB 12V 30 LEDs / meter
  
  Any ideas on how to remove the flickering? Or is my only chance to use a regular RGB stripe instead?

huggy-d1, 2021-07-29:
I recommend a huge audio capacitor across your DC storage (batteries) that provide the 12V used by the WLED controllers and LED strips.
This way, a single item fixes the problems - _assuming_ the problem is due to the voltage source getting pulses on every inverter switching cycle.

It might be the size of a 1 liter bottle if you are drawing 30+ amps @ 12V.

If inverter power and LED power / data lines are bundled together, you could be getting noise from the inverter getting onto the data line.  This can be tested by undoing the bundle and putting 10+ cm of distance between them.  If it stops flickering, you know that is the reason and you can use shielded cable for the power wiring, or stick it inside a metal pipe / conduit, or wrap in metal mesh (or even Al foil for testing).

Kaistee93, 2021-07-29:
@huggy-d1 Thanks for your reply and the helpful hints :)
Unfortunately I don't own such a huge capacitor right now so guess I'll try many small ones and see if that helps. Instead I've got a old 12V car battery - could this work too? The stripe only draws 4Amps in white and max brightness. The Step down converter is of course oversized for it's purpose. 
The wiring is like 
From solar panels goes 5m solar Al wire to the Inverter which includes mppt then 40cm battery copper cable to the battery 24V (=2*12V) and from the battery 40cm loudspeaker wire copper to the first buck converter followed by 5cm loudspeaker wire copper connected to the Relais board AND another buck converter for powering the ESP followed by 3m loudspeaker wire for the stripe and two of the CAT7 schielded data lines made from solid copper to the Vin and ground of NodeMCU V3. 
An extra ground wire is connected from the buck converters input to it's output and the next buck converters input and output. 

Is there a way to check for pulses without an oscilloscope? I don't have one and know nobody who would lend me one.

huggy-d1, 2021-07-29:
Using a 12V car battery is a fantastic voltage smoothing device.  Unfortunately, it will provide 12.x to 14.x volts out, depending on its current charge level.  It will smooth out input the a 12V converter.
As a test though, try powering the pump with the car battery so the pulsing caused by the pump gets isolated by the battery.

Kaistee93, 2021-07-31:
Now it's getting crazy o.O
I've disconnected the cables from the 24v battery pack and instead hooked them up to the 12v battery which had 12.7V and it's still flickering. Disconnecting the Inverter too and the flickering is gone -_-

srg74, 2021-08-08:
First of all serial data cables don't like a twisted pair cables like a CAT-5(6). Use regular 18AWG for data. Second is clearly your inverter generateing interference. You can try to eliminate interference with combination of capacitors near controller power. Use one electrolytic capacitor approx 500uF 35V and ceramic capacitor 0.1uF 50V maybe 10nF. With ceramic capacitor you have to play. And also you can try a ferrite bead on your power feed.

closed 2022-04-16 by Aircoookie