wled/WLED #2104: Control Strips on a Pin Basis

dremerb, 2021-07-29:
The support for multiple strips is amazing! Allows for more lights per ESP.
While we have segments to virtually split strip, the output to all pins currently is the same. How about enabling dedicated controls for every connected strips? Would allow to e. g. have one ESP per room, but with different lighting zones.

huggy-d1, 2021-07-29:
I believe that is working in someone's development fork right now.  You may be able to search the topics on the WLED discourse site and find it, download / install and try it out.

blazoncek, 2021-07-29:
This is indeed in works, very early stages though.

blazoncek, 2021-10-06:
DDP out (which will solve this issue) will be available shortly with next 0.13 beta release.
Closing.

closed 2021-10-06 by blazoncek