wled/WLED #2105: ESP32 (powered by a lipo battery) starts to reboot every few seconds, once battery gets "too empty"

glowingkitty, 2021-07-29:
**Describe the bug**
Noticed both with my TinyPico as well as my TTGO T7 (both ESP32 boards with a lipo charging circuit) that WLED starts to reboot every few seconds, once the lipo battery goes below a certain battery level. More specifically I notice the rainbow animation restarts every few seconds, as if the microcontroller would reboot all the time. I assume the issue is that once the voltage drops below a certain point, WLED has issues and reboots. It definitely is a WLED issue, since I had it now on two different ESP32 boards and more important: when I tested the TTGO T7 a few months ago with my own software (not based on WLED), also running a rainbow animation on the LEDs - I had no such rebooting issue. Instead the LEDs just turned black at some point (off) - which is also the behaviour one would expect.

**WLED version**
 - Board: TinyPico (USB-C) & TTGO T7
 - Version v0.13.0-b2
 - Format self-compiled

Thank you for your help!

huggy-d1, 2021-07-29:
ESP32's have brownout protection built in.  You can search online for ways to prevent the brown-out, or to disable the brown-out restarts you are seeing.  This is not WLED's fault.

blazoncek, 2021-07-29:
Your software may have the CPU sit idle and not use much power. WLED uses a lot of CPU cycles.

glowingkitty, 2021-07-29:
it's an issue I didn't had with my own code which didn't use WLED. There the rainbow animation (created via Adafruit Neopixel library) just turned off once the lipo was empty. Which is also the expected behavior. Therefore it's fair to call it an issue that's related to WLED in combination with an ESP32, not just because of the ESP32.
While we are already at it... In addition I noticed this reboot/restart of the animation happens always exactly once when turning on the ESP32.

glowingkitty, 2021-07-29:
So what's the solution? is there a hardware-based solution or software-based solution (changes I need to make to WLED)?

blazoncek, 2021-07-29:
you can try to add a 1000uF capacitor to 3.3V pin on ESP.

shauneccles, 2021-07-29:
Stick a scope on the power rails and have a look - I bet you it's brownout detection.

glowingkitty, 2021-07-30:
@blazoncek even if one doesn't use the 3.3v pin for the LEDs, but oly for a microphone?

@shauneccles ehm... does a typical multimeter also do the job? just measuring the voltage?

blazoncek, 2021-07-30:
Yes, capacitor (if large enough) may prevent brownouts.
No, the brownout "spikes" may be to short to detect on regular multimeter.

glowingkitty, 2021-07-30:
I assume the V rating for the capacitors is only the maximum voltage they can deal with? so basically any would do for 3.3v? and how come I should connect it to 3.3v, not the 5v pin, where the LEDs are connected to?
@blazoncek

blazoncek, 2021-07-30:
> and how come I should connect it to 3.3v, not the 5v pin, where the LEDs are connected to?

3.3V pin is for powering ESP and you want to provide ESP with adequate power (no voltage drops). So you attach capacitor to a power input pin to ESP.
Capacitor V rating is best to have 2x used voltage, so in this case I would recommend 10V or at least 6V.

glowingkitty, 2021-07-30:
but i am not using the 3.3v pin to power the ESP32... but the lipo connector on the back of the TTGO T7, with a lipo battery (3.7v).

blazoncek, 2021-07-30:
So? ESP is powered by 3.3V, it (TTGO) has an onboard 3.7-5V to 3.3V converter so that you don't fry it when powering from USB or other sources with voltage >3.3V.
You should learn a bit about electronic circuits I guess. 😉

glowingkitty, 2021-07-30:
that's why I build things with electronics and ask questions, to learn more about electronic circuits ;)

Let's assume for a second that the 1000uF capacitor would fix the brownout issues. Just searched online for such a capacitor 3d model, to get a feeling for the size of it... and... it would definitely be too large for my wearable... would a smaller capacitor also do the job?
[image]

shauneccles, 2021-07-30:
Let's confirm its the brownout detector - https://www.robmiles.com/journal/2020/1/20/disabling-the-esp32-brownout-detector

Putting a scope on would let you know more, it's possible a smaller capacitor will be enough but you'd have to proto it.

blazoncek, 2021-07-31:
The WLED startup code already has a possibility to disable brownout detector.
Just use appropriate #define.

You may experience unexpected behaviour though if voltage fluctuates too much.

pbolduc, 2021-08-20:
WLED_DISABLE_BROWNOUT_DET has to be undefined...

https://github.com/Aircoookie/WLED/blob/3ad336a1eb42d7023bb1c548b66e50b4cafbd190/wled00/wled.cpp#L254

closed 2022-04-16 by Aircoookie