wled/WLED #2106: preset corruption after wlan disconnect

poelzi, 2021-07-29:
**Describe the bug**
A clear and concise description of what the bug is. Please quickly search existing issues first!
After wled looses it's wlan connection, saving a preset causes preset corruption.
It causes entries like this:
```
{"n":"pastaleri","win":"Please refresh the page to see this newly saved command."}}
```

**To Reproduce**
Open UI. Kill the WLAN connection of the ESP. When the ESP is back online, continue configure the settings and save as a
new preset. This cause the "win" attribute to contain above garbage.
I think I used chromium 

**Expected behavior**
A clear and concise description of what you expected to happen.

**WLED version**
 - Board: nodemcu esp32
 - Version 0.12.0
 - Format Binary,

huggy-d1, 2021-07-30:
Have you tried the current released version to see if the bug you found has already been corrected, and therefore no longer an issue?  V0.13.x

blazoncek, 2021-07-30:
This should only happen if you save twice, since on the save, current (saved) preset is overwritten with the text you provided.
You should refresh your browser on preset save, to see the newly saved preset.

This is not related to killing WiFi.

closed 2021-10-06 by blazoncek