wled/WLED #2107: ArtNet(or E1.31) output

jonslilbro, 2021-07-30:
I love the effects generator that is built into WLED and the ability to quickly make custom effects for mainly difference situations and colors without having to have a heavy knowledge of the software. There are many times I find myself having to use a larger controller for pixel control(say a Falcon board) just because of layout and outputs and then I'm stuck having to use something like Madrix to program and run the controller. I'd really love if there was a way to output ArtNet or sACN to a different controller and just let WLED be the effects generator.

Any chance of this getting integrated at some point?

huggy-d1, 2021-07-30:
WLED accepts input from ArtNet / DMX if that is what you are asking.  Then you can do what you wish.  If you want WLED to do what it does, you can have the special software thing issue CURL commands (use the JSON or HTTP API) to set effects and related values at the proper times in the "show".

jonslilbro, 2021-07-30:
> WLED accepts input from ArtNet / DMX if that is what you are asking. Then you can do what you wish. If you want WLED to do what it does, you can have the special software thing issue CURL commands (use the JSON or HTTP API) to set effects and related values at the proper times in the "show".

I'm looking to WLED to do ArtNet OUTPUT not INPUT. I'd like to have WLED just be the broadcaster in certain situations and not the receiver.

huggy-d1, 2021-07-30:
You mean to send out a sync signal, which it already can do?  Or, you mean WLED become a show coordinator?  I think that kind of scope creep is enough to cause anyone to go, oh no way is this the right tool for that.  Just my opinion of course.  Others might look at it as a challenge and actually try it, or analyze it enough to respond it's possible or not possible for an ESP32.  I cannot imagine an ESP8266 would have enough resources to do that.

jonslilbro, 2021-07-30:
Not a sync signal, but an actual ArtNet(or sACN) signal. It's already creating the effects so can handle processing that just fine and we know it's capable of receiving ArtNet and processing the input signal out to the LEDs it's controlling so it doesn't seem like too crazy of a load to just send instead of receive. I'd like to be able to setup a WLED to have say 1200 pixels and send ArtNet out to a Falcon F16v3 that I have the 1200 pixels hooked up to via differential receivers and be able to control the 1200 pixels from WLED using it's built in effects generator.  Definitely would want to stick to ESP32 for it, especially to allow more FPS or more pixels, but when controlling say 1200(or more) pixels that are spread across a few different areas, using something like 3 different areas in a yard, using a controller like a Falcon or Kulp or SanDevices controller makes the most sense, but then I'm stuck having to use something that is preprogrammed from like XLights or Madrix where I would love to be able to just jump into the webpage for the WLED setup and make an on the fly change to colors easily.

blazoncek, 2021-07-30:
I guess that would be too resource intensive for ESP.

pbolduc, 2021-08-20:
I was thinking about this issue/question this morning. One challenge for having WLED directly generate an ArtNet(or sACN) signal is knowing how much data to produce.  How many universes should it send out? How many channels for each one of those universes? Also some effects like chase or fireworks 1d may never look great. I was thinking a separate "WLED to sACN" bridge app could be developer by _someone_ that takes the UDP sync signal from the master WLED instance, and converts the change (effect, pallet, brightness, intensity, etc) to corresponding network signals to each controller. This other app would need some configuration to know the IP addresses of each of your other controllers, the universe and the respective number of channels.  If one just wanted to replicate the WLED effects on a falcon/kulk, it could also be possible (with someone writing a program) to process an effect into a fseq file that could be loaded onto FPP.  This latter approach is less flexible but would eliminate issues with realtime updates.

I do think this kind of feature goes beyond what we could expect WLED to run on a microcontroller.

blazoncek, 2021-09-24:
Me an @pbolduc implemented DDP out from WLED.
Be warrned though that ESP32 and especially ESP8266 do not have resources to handle more than a few hundred LEDs.
Support for E1.31 and ArtNet is planned for future.

The code is currently available in my (dev) or @pbolduc 's (feature/upd-ddp-send) fork and testing it is required before it can be merged to master.

blazoncek, 2021-10-06:
E1.31 and ArtNet out are in the works. DDP out will be available shortly in next beta release.
Closing.

closed 2021-10-06 by blazoncek