wled/WLED #2108: Both .13 betas cause loop reboot on M5stack atom lite

Kodekat84, 2021-07-30:
**Describe the bug**
Flashed to m5Stack atom and endless reboot occurs.

**To Reproduce**
Flashed with ESPHome Flasher 1.1.0

**Expected behavior**
Module boots and wifi access point pop-up to configure.

**WLED version**
 - Board: ESP32-PICO
 - Version 0.13.0-b2 \ 0.13.0-b0
 - Format Binary

**Additional context**
Anything else you'd like to say about the problem?

Thank you for your help!

huggy-d1, 2021-07-30:
Has anyone had success with the ESP32-PICO's yet?  I have only seen problems.  Did you search the discourse site and existing issues for PICO to see if anyone has been successful with that device yet?
This discourse thread may help: https://wled.discourse.group/t/support-for-esp32-pico/265/14

blazoncek, 2021-07-30:
@srg74 has had success.
He may share info on how on WLED Discord.

phluther, 2021-08-04:
The available binary for the ESP32 is not compiled for the ESP32 Pico that is used in the Atom Matrix and Atom Lite. If you want to install on the Atom, you need to compile WLED from source using platformio. You will need to uncomment "default_envs = m5atom" in "platformio.ini".

Aircoookie, 2021-08-06:
Indeed I can confirm the `m5atom` environment works, while the binaries (which are built with the `esp32dev` environment) don't.
I will try to check whether it is possible to make the ESP32 Pico compatible with the standard binary again.

srg74, 2021-08-06:
I can confirm that latest master is working with ESP32-pico-kit.
[image]

blacksanta69, 2021-09-08:
> I can confirm that latest master is working with ESP32-pico-kit. WLED 0.13.0-b2 (build 2107230)

I can successfully install 0.13.0-b2 or any v13 beta _only_ on-top of a successful v12.0 write and setting some default configs - e.g. wifi settings. So, I'm guessing this issue is bound in some way cfg.json or the wifi settings not being set and it triggering an error (or I'm missing some other step). I definitely _cannot_ get this running after erasing and then writing v0.13.0-b2 bin with platform.io or esptool; I get the looping error (Guru Meditation Error: Core 1 panic'ed (IllegalInstruction))

Hope this helps.

For me, the current workaround is to install v12, set the wifi settings and then install any v13b

closed 2021-12-18 by blazoncek

jcam, 2023-08-19:
This is the still the case, and is also a problem on the newer AtomS3 Lite running either 0.14 or MM. Have to do a custom build to avoid WLED triggering a FLASH_CS reset on bootup