wled/WLED #2111: Possible to support Philips Wiz lights?

sofakng, 2021-08-01:
**Is your feature request related to a problem? Please describe.**
Philips Wiz lights are extremely cheap compared to Philips Hue and it would be nice to see them supported.

**Describe the solution you'd like**
Support for Philips Wiz bulbs.

**Describe alternatives you've considered**
n/a

**Additional context**
There are Python libraries for controlling Philips Wiz and I think the protocol has been (mostly?) reverse engineered.

huggy-d1, 2021-08-02:
I think you can find online resources for hacking into the wiz lights, loading WLED to it (or Tasmota), and from there, you have access to it via wifi for loading whatever firmware will fit inside available memory.

probonopd, 2021-08-29:
> I think you can find online resources for hacking into the wiz lights

Is this possible without physically opening the Wiz devices?

ChuckMash, 2021-09-08:
I've been able to get WLED running on a WiZ LED Strip & Controller, just takes a bit to deal with the single core ESP32 variant.
https://github.com/Aircoookie/WLED/issues/2171

I've also been able to get WLED loaded onto the bulbs, however, the process of gaining physical access to the ESP is decently destructive and leaves the bulb in an extremely impractical state for use.

I'm still examining other options for non-destructive modification for the bulbs, but it doesn't look good at this point.

ChuckMash, 2023-11-20:
By the way, and for anyone else stumbling to this issue, WiZ Lights have now been supported by WLED for a little while now with a usermod.
https://github.com/Aircoookie/WLED/tree/main/usermods/wizlights

closed 2025-11-09 by netmindz