wled/WLED #2112: sync not synchronizing all elements accurately and finding weleds but not adding to list in app

ThatBigPrint, 2021-08-01:
**Describe the bug**
First issue,

when I have multiple wled modules one or two of them don't sync properly you then need to resubmit the change then it applies this is intermittent.

second issue,
 
the wled app finds the uniquely named modules but after you click the tick it only populates a few but if you manually type in IP address it populates as it should ?

**To Reproduce**
first issue you just need multiple items on one sync id for instance I have 11 

second issue again multiple items on same network I have 15 in this instance 

**Expected behavior**
all wled to sync reliably

app to be able to find nodes every time and populate 

**WLED version**
 - Board: [e.g. ESP32 dev]
 - Version [e.g. 0.13.0, beta] although other nodes I'm having the same issue with are on stable old releases 
 - Format [e.g. Binary, self-compiled] both 

**Additional context**
Anything else you'd like to say about the problem?

Thank you for your help!

huggy-d1, 2021-08-02:
How many wifi access points do you have?  Are all WLED installs connect to the same wifi AP?  The sync protocol does not route as far as I know since it's a simple UDP broadcast.  WLED simply sends out to whoever is capable of listening, and is listening.  The networking configuration determines whether they are capable of listening.

ThatBigPrint, 2021-08-03:
they are all on the same network they only produce ap if they can't find the known SSID ? so far they are all on the network and not complaining, the issue lies in reliable sync and pulling into the app on both platforms... they initially worked but since I deleted the list and re searched only one or two come back and its always the same one or two

ThatBigPrint, 2021-08-06:
a bit more info on this... the Host light seems to drop out for whatever reason, and then comes back around it happens on all of the elements whichever is the host I'm even considering making a hidden dummy hoast? anyway use this link to see a video of the issues https://youtu.be/eC7rsYlEwow

thanks
chris

blazoncek, 2021-08-23:
WLED Sync uses UDP. That is in itself enough said about reliability on WiFi.
You can slightly improve the situation by telling WLED to transmit sync message twice. Use Settings page.

keck9939, 2022-12-04:
The sync feature relies on mDNS and UDP broadcasting. Depending on your networking equipment and topology, these two features may or may not be enabled/blocked. I found with the default settings on my network, that enumerating the WLEDS on the network was unreliable and that nodes would periodically drop off the network. After enabling mDNS and making sure UDP broadcasts would get through, everything started to work reliably.

blazoncek, 2023-01-25:
Mitigation added in #2830

closed 2023-01-25 by blazoncek