wled/WLED #2115: effects work on PC Via bowser but via phone browser or app cant change

FigJam23, 2021-08-04:
**Describe the bug**
Unable to change effects via phone browser or 
WLED app but works via browser on PC Fine

**To Reproduce**
Use using phone either android or apple dosent work via PC changes fine

**Expected behavior**
Should change effects via phone browser or app but only works on PC

**WLED version**
 -  ESP32 dev kit v1]
 -
 - WLED_0.12.0_ESP32.bin

Thank you for your help!

huggy-d1, 2021-08-04:
Have you tried the latest firmware which probably addressed the bug you are experiencing?  v0.13.x

Aircoookie, 2021-08-05:
What is your phone model, OS version and browser? :)

FigJam23, 2021-08-06:
> Have you tried the latest firmware which probably addressed the bug you are experiencing? v0.13.x

Yeah was running latest version of bats downgraded to latest stable with same issue. Strangely after a few reboots it now seems to work as it should

FigJam23, 2021-08-06:
> What is your phone model, OS version and browser? :)

Seems to be working now but just in case it helps devices that issues was not working on was a Google pixel 2xl and a iphone 8 running latest os

closed 2021-08-06 by FigJam23