wled/WLED #2116: Energy measurement addition to WLED

fribse, 2021-08-04:
**Is your feature request related to a problem? Please describe.**
Currently there are no insights into the power draw from the WLED installations. It would be a great addition to have that added.

**Describe the solution you'd like**
Something that can be fetched via the API for the new Energy monitoring features of Home Assistant (support for long term statistics: https://developers.home-assistant.io/docs/core/entity/sensor/#long-term-statistics)

**Describe alternatives you've considered**
I could add a measuring plug in front of all the WLED devices, but it seems clumsy as the device should be able to calculate the consumption of the devices, and then with a variable as baseline for 'passive' usage, then it would be honky dory.

**Additional context**
Add any other context or screenshots about the feature request here.

Thank you for your ideas for making WLED better!

huggy-d1, 2021-08-04:
You can configure current maximum.  You can configure # of LEDs.   You can configure the type of LED (WS281x for example).  There is a considerable amount of code dedicated to the very thing you state is missing.  In a properly configured WLED install, it does a fair estimate (erring on the safe side) of current usage and will limit current consumption by reducing brightness if there is a configured maximum current setting.

sansillusion, 2021-08-04:
All that information is already provided.
!
[Screenshot_20210804-182733](https://user-images.githubusercontent.com/29512074/128263520-f5f25d90-ebf7-4378-a2a9-9656ccf855a5.png)
I get it all in my Home Assistant integration, and it can be viewed in the browser when clicking info button at the top.

fribse, 2021-08-06:
> In a properly configured WLED install...

Are you serious? In a properly formulated answer it would not be written condesending!
I do know how to set up WLED, I've been a long time fan of the project, I don't know how to code, but at least I can send some money once in a while and support it that way. Your reply only tells me one thing, and that is not a nice sentiment about your attitude.

When a suggestion is made, it is not an attack on something that is wrong, it is a wish for something to get better. Behaving the other way will allianate people and drive the project to the ground, 

And no, @huggy-d1 and @sansillusion you are not right it does not give out the proper values, there are no W or Wh output, which is needed for the energy measurement in HA.
There are speific needs for that, it needs an accumulated daily consumption, with a reset timestamp.
It can probably be built in the integration as well but it is very complex and has been rejected by Frenck: https://github.com/home-assistant/core/issues/53972#issuecomment-892787046.

fishbone-git, 2021-08-06:
Add the following to your Home assistant sensors config, with some changes to suit your setup.  I use 12V, you may use 5 and I also have a relay to cut power to the LED strip, which is why I have 4W if off for the digQuad only.  I get where Frenck is coming from; it makes little sense to add in the integration if HA config can do it. WLED has the critical info (namely estimated current) needed to make it work with the new energy tracking.  Even a usermod is probably more complicated than the config below. I've compared with a power measurement socket, and the result is pretty close (just required me adding the when off check).  At max brightness, it's dead on.  Less then 100% it can vay by a couple of W, but for me that's fine and the WLED software isn't going to be able to do much better without a hardware requirement.

- platform: template
  sensors:
    my_wled_power:
      friendly_name: 'WLED Power'
      unit_of_measurement: 'W'
      value_template: >- 
          {% if is_state('light.wled_lights','on') %}
            {{ states('sensor.wled_Lights_estimated_current')|float * 12/1000 }}
          {% else %}
            {{ 4 }}
          {% endif %}
- platform: integration
  source: sensor.my_wled_power
  name: WLED Consumed Energy
  round: 3
  unit_prefix: k

huggy-d1, 2021-08-06:
@fribse I have improperly configured WLED more than once.  I realized it when I blew a fuse, or brightness was being limited _for now reason_ other than my oversight.  I wish I was as good as you at configuring WLED.

I think I understand now what you mean by the issue posting.  I apologize for completely misunderstanding what you were asking for.

I obviously thought you were asking for a way to measure energy, not have WLED compute, store, display, and make available via API.

Now I know you want WLED to get energy computation added.  What input fields do you think would be required for this additional computation?  Something like this _for each LED pin_:

- *LED strip voltage
- current per LED (instead of a default value)
- LED count

Where: * new user configuration entry

Make sense?

Aircoookie, 2021-08-06:
Hey!
This is a great suggestion, thank you!
Getting the energy usage is as simple as multiplying the power (either the already implemented estimate, or optionally) over a period of time. To accurately measure/estimate the consumption, this could even be done every animation frame so that all brightness fluctuations due to the effect mode are honored.

My question would be which value(s) would need to be made available in the API so that integrations (for example HomeAssistant) can implements long term energy monitoring. Unit is pretty straightforward, either watt-seconds or watt-hours would likely make the most sense. The problem is the timeframe: Do we want the energy used since the last query? Last 5 seconds? Last minute? Since boot? This is still a bit unclear to me. Maybe @Frenck could help me determine what time frame would be most suitable.

Additionally, output of the current power in W could be useful (currently it is only mA). We can not measure the voltage without extra hardware, that is the reason the current estimation is current only, not wattage. However, the voltage could be estimated to be either 5V or 12V, which would allow for a semi-accurate current wattage reading as well.

fribse, 2021-08-06:
Hey @huggy-d1 
No problem, we're all good.

@fishbone-git 
That would absolutely be a way to circumvent the challenge, just a bit backwards imho, and it's quite a bit of work here with the amount of wled implementations I have (and not for the average user I guess) but until implemented it could be a temporary solution. I completely agree that wled will not be able to make any more precise readouts. To have it more precise one would have to add something like a HLW8012, and I think that's a lot of work for very little benefit.

@Aircoookie I think it would be sufficient to just allow people to enter the voltage in the LED setup, and then comment on the field that it is only used for calculations.
It would probably be necessary to have an 'idle' energy usage as well, which could of course be a list of suggested values for known configs, or one could measure it oneself and just enter a value.

The current usage I've set up for my ESP home devices is a 'accumulated' with a reset time stamp.
So every night it resets, and until that it just sums up.
So it's an ongoing summation, I guess it would be more than enough that it sums every minute or on a state change, I'm not sure if that is clear?
so a run like
idle - idle - idle (updated every 60s), led state change, new value added to the accumulated usage every 60s, etc.
I don't think that an update more often than 60s makes any sense, maybe even lower.
The energy readouts I've seen are shown in Wh, and then shown in kWh if necessary in HA.
There are some explanations on the requirements for the statistics (which the energy monitoring used) explained here: https://developers.home-assistant.io/docs/core/entity/sensor/#long-term-statistics

Aircoookie, 2021-08-06:
Ah, thank you, I had taken a look at the page but not realized that `Metered entities` is a sub-point of Long term statistics.
In that case, just accumulating the energy usage starting from boot is likely the simplest way.
I don't think it makes sense to provide a `last_reset` timestamp in the API, since the WLED device might not have an accurate time source like NTP enabled. The uptime since the last reset is accessible in the API though (`info.uptime`) and the integration can simply subtract that from the current timestamp to get `last_reset`.

fribse, 2021-08-07:
Yes, that sounds like a good solution, and even doable in the example from @fishbone-git .

fribse, 2021-08-07:
The uptime sensor is apparently disabled per default :-) But I've enabled it now.
I don't understand the output though if the wled doesn't have time, but that could be the integration changing it of course???
{{ states('sensor.adrians_skrivebord_uptime') }} gives out
2021-07-11T10:47:30+00:00
And that is not an uptime, that is a boottime?

fribse, 2021-08-07:
@fishbone-git Have you added the sensors to the HA energy system? Mine doesn't show up, and I'm unsure why, as far as I can see the requirements for the 'last reset' and state_class: measurement are there?
[image]
(So far I've left your 4W standby as is)

fishbone-git, 2021-08-07:
Another workaround is needed to make the entities compatible with the new
Energy platform. It's not just WLED in this case (my own setup, I had to
even fix everything) Any entity that you want to show in the device list
must have last_reset, device_class and state_class. You can customize each
in HA, or just do it globally for all energy sensors (that follow the
naming pattern).

homeassistant: customize_glob: sensor.*_energy: last_reset: '1970-01-01T00:
00:00+00:00' device_class: energy state_class: measurement

fribse, 2021-08-07:
Ahh, it's the device_class I forgot :-)

fribse, 2021-08-09:
Oh, btw. @fishbone-git I had to modify the energy sensors for wled, and the global customization, as that also hit the other energy sensors giving me negative values :-) So I added wled in front of energy, so it's specific for the wled sensors.

frenck, 2021-08-09:
So, kinda shocked a bit by the whole thread here.

Last reset and all other stuff isn't really a thing WLED needs to worry about. The reason the current integration cannot adopt energy measurements for WLED is simply that it doesn't provide energy.

So, what information does WLED provide right now? Well, it provides `current` (`I`), but in order to calculate the energy usage, one needs to know `power` (`P`).

`P = U * I`

We are missing Voltage in the API (`U`) in order to be able to know power usage. After that, one could integrate the energy usage from that. However, it would not provide energy, but only power, it would still require a manual setup to integrate the energy usage from power.

So, ideally, WLED should also provide energy usage (e.g., `Wh` or `kWh`). Personally, I think it makes the most sense, to provide this value since the last boot. This prevents extensive writing/updates to the flash memory (wearing it down).

---

From the Home Assistant side of things, we can detect a drop/re-connect and adjust all internal Home Assistant things (like last_reset) and such. All these things are not relevant to WLED.

---

TL;DR:

Would be cool if WLED could provide voltage, power & energy information.

sansillusion, 2021-08-09:
There is multiple ways to do it in the end. It's also possible to do it all
in the HA integration with no modification to wled. There could be a
seeting for voltage in the integration configuration (for each wled
insurance) and the integration could calculate it all on it's own. I have
no idea who is in charge of the HA integration but in every circonstances
it will need modification.

frenck, 2021-08-09:
> It's also possible to do it all in the HA integration with no modification to wled.

That is incorrect. Home Assistant doesn't allow for "virtual" / generated sensors, which is against the architecture.

sansillusion, 2021-08-09:
Oh yea, I kinda mixed integration and add-ons. So then we need wled to
provide the data I guess. Would have been nice to preseve the limited
ressources of our esp's. I am still surprised at what wled can do with them.

Aircoookie, 2021-08-09:
@frenck thank you for your perspective on this :)

I fully agree, providing both the current power and the energy usage since boot via API is the best course of action. (only integrating over e.g. a minute is bad in case the connection is lost, and storing it beyond reboots is complex and overkill, as you said)

What I am torn about is voltage, however. It is on my agenda to adapt the power estimation code to properly work with multiple pin outputs, e.g. allow a user to configure e.g. one string with 50 5V and another with 100 12V LEDs. Could calculate an average from that, but it likely doesn't make much sense to do so. In this (arguably edge case), `P != U * I`, at least in the values reported by the API. Alternatively, we could do two arrays with the voltage and current for each bus respectively, that might be useful, might be overkill though... 
The user definitely needs to be asked the voltage anyways (otherwise WLED couldn't calculate the power and thus energy),
I'm just wondering whether providing that value in the API would be beneficial.

frenck, 2021-08-09:
> I'm just wondering whether providing that value in the API would be beneficial.

IMHO it's not...

I think anybody truly looking into energy monitoring, should just use a wall plug that does just that. It will provide actual data, instead of estimations.

fribse, 2021-08-10:
@frenck I don't agree, some of my WLED installations can't have plugs because of different constraints. I do have a bunch of BlitzWolfs BW-SHP6,7,9 around doing all sorts of monitoring, but it's not doable in all situations, and have them in a good quality is not a couple of bucks, it's more like 15-20 pr. piece.
The best would of course be to have an integration to HLW8012 and have something like diguno / digquad support that, but I think that would require even more code.

frenck, 2021-08-10:
@fribse Nothing is stopping you from writing a sensor that makes an estimation...

fribse, 2021-08-11:
@frenck no, that's true, but that might be a bit steep for others :-D
It's a suggestion not a requirement, I think it would be a wonderfull addition, so I'm hoping Aircookie will add the functionallity, but it's not up to me if it's doable spacewize or anything.

lsaadeh, 2022-08-28:
Thanks for raising this issue @fribse, I came here with the same interest in this feature and to bump this issue again. 

While I understand the points everyone has made about accurate energy monitoring, a lot of us we're just interested in a close approximation to get an idea of how much energy our leds are consuming over a period of time.

I think it would be a good idea as spoken about to provide the kwh to ha to not store any long-term data on the device and provide the voltage through the UI as a start. Ultimately a lot of us are looking to add our wled devices as individual devices in the energy dashboard. 

Due to the addition of energy monitoring in ha dashboard, this has become a quite commonly requested feature of energy-consuming devices and would be a fantastic addition to wled!

softhack007, 2022-08-28:
> So, ideally, WLED should also provide energy usage (e.g., Wh or kWh). Personally, I think it makes the most sense, to provide this value since the last boot. 

As several people said already, today WLED computers an estimate of  the _momentary_ current `I(t)`. To get the power  or energy consumption, the first step would be to integrate this current over time. 

If WLED would just uses it's own estimated current, the result might become extremely inaccurate. In numerics, we call that "computing street numbers", meaning that the results you will get are _farway_ from reality.

Why ?
A) a typical LED strip effect runs with 40fps, so there are 40 "current measurement" to be integrated per second. 
B) Inaccurate estimations tend to "pile up" when integrating over time. For example, if your estimation is always 100mA too high, it means that after four hours the estimated energy / power consumption is wrong by up to 57.600 watts / 14.4kwh.

To avoid such numerical disasters, it might help to first compute a "per-minute average current" and then integrate this value once per minute.  

Instead of using a "rolling average" exponential (IIR) filter, I would suggest to sum up current samples over a fixed time slice, devide the result by number of samples taken, then integrate this "minute average". Start next cycle without considering the previous measurements. This is basically a low-pass FIR filter that avoids the very nasty behaviour of IIR filters.

blazoncek, 2022-08-28:
All this assuming you **know** how much each LED consumes. Which yo don't. You can use datasheet to get an average but THAT may still be off. And it may even have a nonlinear reponse.

Best to give up on the idea IMO.

chunkysteveo, 2022-11-25:
Have just read through all this and tested installing PowerCalc using HACS - https://github.com/bramstroker/homeassistant-powercalc and it was an absolute dream to add in power and energy for WLED. 

After install and reboot - all my WLED lights came up as new PowerCalc entities in the Integrations page - add them, configure the WLED settings of Voltage and power factor (if adjusting it), and they started feeding into the Energy dashboard after adding them in there.

Will be good to see with strips (Christmas Lights) are using the most power, and now to get some (rough) good estimates on the amount of kWh spent!

smish66, 2022-12-07:
Hey @chunkysteveo , I'm curious on how you were able to get your WLED lights to show up? I have a few WLED's in my integration, but I do not see any automatically show up in the PowerCalc integration. Interesting since the PowerCalc integration doesn't list WLED as a supported light. Just wondering if you did anything to get them to show up?

chunkysteveo, 2022-12-08:
Hi @smish66 ! Using latest versions of WLED, HA and PowerCalc that I installed using HACs - when the new i stance of WLED is discovered in HA, I add it and configure it. Reboot HA, and PowerCalc automatically detects WLED as a new entity to add power and energy to (in integrations tab). Add and configure those and PowerCalc adds the sensors to the WLED device in HA.

PowerCalc allows you to add grouped sensors too (manual +integration), so I added up all my WLED sensors. And added a template sensor for the total cost too using energy and my electricity £/kWh..... it's reassuring that almost 1500 LEDs has only cost me about £2 to light so far in December!

WLED is a profile in PowerCalc, and it detects WLED automatically with the latest versions. I didn't configure or change anything with PowerCalc after installing from HACs. It does seem to need a reboot after adding WLED for it to detect them?! Are you adding new WLED instances? If not - try deleting the WLED device, and re-adding again (another reboot after delection should let HA spot it automatically).

When PowerCalc spots them and you add to configure the device, it asks for voltage and power factor. It's either 5v or 12v depending on your LED setup, and I left power factor as 0.9.

Let me know how you get on...!!

chunkysteveo, 2022-12-08:
[image]

£1.46 actually! 👍🏻

smish66, 2022-12-09:
Thanks @chunkysteveo , I will give that I try. Thanks for the screenshots as well.

softhack007, 2023-01-06:
Similar requests and discussions:
#1572 
#1657

gothdawid, 2025-01-24:
> Add the following to your Home assistant sensors config, with some changes to suit your setup. I use 12V, you may use 5 and I also have a relay to cut power to the LED strip, which is why I have 4W [...]

I have improved your code a bit.
Includes when the device is not available.

```YAML
sensor:
  - platform: template
    sensors:
      wled_studzik_estimated_power:
        friendly_name: "Obliczona Moc WLED1 (Studzik)"
        unit_of_measurement: "W"
        device_class: power # Klasa dla mocy (W)
        value_template: >-
          {% if is_state('light.wled_studzik_2','on') %}
          {{ states('sensor.wled_studzik_estimated_current') | float * 12 / 1000 }}
          {% elif is_state('light.wled_studzik_2','off') %}
          {{ 1.1 }}
          {% else %}
          {{ 0 }}
          {% endif %}

  - platform: integration
    source: sensor.wled_studzik_estimated_power
    name: Zuzycie Energii WLED 1 (Studzik)
    unit_prefix: k
    unit_time: h
    round: 3
    method: left
```