wled/WLED #2117: Hardware Setup is not saved

Gilias-Ger, 2021-08-04:
I updated my WLED Lamp from 9.1 to 12.0 since the Lamp has stopped working.

WLED does not save the Configuration of the LED. I have carried out the "Hardware setup" under "LED & Hardware setup" and clicked on Save. After calling up the Page again, the data entered are not saved

I downgrade again to "WLED_0.9.1_ESP8266_ws2801" everything works again.

My System:
 - Board: Wemos D1
 - Version WLED_0.9.1_ESP8266_ws2801 / WLED_0.12.0_ESP8266
 - Format Binary
 - LED WS2801 RBG

What is actually queried for "Data pin" and "Clk", the GPIO or the ESP number?

I've already tried both

GPIO: 0/2
ESP pin: 18/17

what is right about it?

Gilias-Ger, 2021-08-06:
I have now tested Version "0.13.0-b2" with exactly the same Problem. The Hardware Settings are not saved :(

blazoncek, 2021-08-06:
What do you consider "hardware settings"?
If it is just LED pins, there was (is?) a bug preventing successful save for APA102/WS2801 chipsets. The workaround is to change LED type & save, then correctlty set type, enter pins and save again.

Gilias-Ger, 2021-08-06:
> What do you consider "hardware settings"?
> If it is just LED pins, there was (is?) a bug preventing successful save for APA102/WS2801 chipsets. The workaround is to change LED type & save, then correctlty set type, enter pins and save again.

Exactly that is not possible!

I change the Type of LED from "WS281x" to "WS2801" and press save and after I go back to the Settings, WS281x is again in LED Outputs...

If I switch from "WS281x" to "WS2801", new settings are also visible. Data and CLK what must be entered there?

Data is D3
CLK is D4

with me.

blazoncek, 2021-08-06:
GPIO pins for data and clock as wired on your setup.

Gilias-Ger, 2021-08-06:
The Main Problem is that he does not Save what I enter

blazoncek, 2021-08-06:
Try with /edit page.

Gilias-Ger, 2021-08-06:
What do I have to Change?

{"rev":[1,0],"vid":2107100,"id":{"mdns":"wled-6d65c1","name":"Lampe-WLED","inv":"Light"},"nw":{"ins":[{"ssid":"XXXX","pskl":19,"ip":[0,0,0,0],"gw":[0,0,0,0],"sn":[255,255,255,0]}]},"ap":{"ssid":"WLED-AP","pskl":8,"chan":1,"hide":0,"behav":0,"ip":[4,3,2,1]},"wifi":{"sleep":true},"hw":{"led":{"total":36,"maxpwr":850,"ledma":55,"rgbwm":3,"ins":[]},"btn":{"max":2,"ins":[{"type":2,"pin":[0],"macros":[0,0,0]},{"type":0,"pin":[-1],"macros":[0,0,0]}],"tt":32,"mqtt":false},"ir":{"pin":4,"type":0},"relay":{"pin":12,"rev":false}},"light":{"scale-bri":100,"pal-mode":0,"gc":{"bri":1,"col":2.8},"tr":{"mode":true,"dur":7,"pal":0},"nl":{"mode":1,"dur":60,"tbri":0,"macro":0}},"def":{"ps":0,"on":true,"bri":128},"if":{"sync":{"port0":21324,"port1":65506,"recv":{"bri":true,"col":true,"fx":true},"send":{"dir":false,"btn":false,"va":false,"hue":true,"macro":false,"twice":false}},"nodes":{"list":true,"bcast":true},"live":{"en":true,"port":5568,"mc":false,"dmx":{"uni":1,"seqskip":false,"addr":1,"mode":4},"timeout":25,"maxbri":false,"no-gc":true,"offset":0},"va":{"alexa":false,"macros":[0,0]},"blynk":{"token":"","host":"blynk-cloud.com","port":80},"mqtt":{"en":false,"broker":"","port":1883,"user":"","pskl":0,"cid":"WLED-6d65c1","topics":{"device":"wled/6d65c1","group":"wled/all"}},"hue":{"en":false,"id":1,"iv":25,"recv":{"on":true,"bri":true,"col":true},"ip":[192,168,178,0]},"ntp":{"en":false,"host":"0.wled.pool.ntp.org","tz":0,"offset":0,"ampm":false,"ln":0,"lt":0}},"ol":{"clock":0,"cntdwn":false,"min":0,"max":29,"o12pix":0,"o5m":false,"osec":false},"timers":{"cntdwn":{"goal":[20,1,1,0,0,0],"macro":0},"ins":[]},"ota":{"lock":false,"lock-wifi":false,"pskl":7,"aota":true},"um":{}}

blazoncek, 2021-08-06:
Looks like you don't have any busses defined.
Look for "hw" and in it "ins". There is an empty array.
You need to create one bus and save it.

blazoncek, 2021-08-06:
Also you may want to try latest 0.13 beta.

Gilias-Ger, 2021-08-07:
I have the last WLED Version now on the WEMOD D1. I have Updated from 9.1 to 13.0 using the "Manual OTA Update" Option. Should all Data remain stored or not?

[image]

> Anscheinend sind keine Busse definiert.
> Suchen Sie nach "hw" und darin "ins". Es gibt ein leeres Array.
> Sie müssen einen Bus erstellen und speichern.

Und was muss ich da eintragen?

mattias1989, 2021-08-07:
I have the same problem, I assembled a lamp with version 12 and it did not work with the IR control. Down to version 11 and if it worked but did not recover the configuration when the chip was reset and the IR control was deactivated. Now install version 10 that works correctly.

blazoncek, 2021-08-08:
Since version 0.12 every pin is configurable. Some also in usermods.
If some feature is not working it must be your setup (old configuration or wiring) that prevents it from working since everything has been tested by many beta testers.

The best way to check if your configuration is causing trouble is factory reset or re-flashing using serial interface and erasing flash memory. If you do not want to loose your configuration you can backup `cfg.json` and `presets.json` using /edit page.

After you have a clean setup try to configure the feature that is not working according to your HW setup. When it is working write down the settings and restore cfg.json and presets.json from backup then correct the settings for your feature.

Gilias-Ger, 2021-08-08:
I have now completely deleted the Wemos D1 and replayed the "0.13.0-b2".

Now he has finally saved the Settings and the Lamp works.

Thanks for your help!

Unfortunately I can't say where the mistake was...

closed 2021-08-08 by Gilias-Ger

blazoncek, 2021-08-08:
👍