wled/WLED #2118: ArtNet wifi node to DMX output: why?

djmuerteitm, 2021-08-05:
Hi!

Can I use WLED as wifi ArtNet node with DMX output through MAX485 board, like @mtongz Code? https://github.com/mtongnz/ESP8266_ArtNetNode_v2

I have a analog RGB LED strip controlled by DMX RGB receiver. I can drive it with Home Assistant ArtNet integration, using a Wemos D1 (lite or pro) as ArtNet wifi node (Mtongz code) and works well, but color selected sometimos jumps randomly  (i hope might be caused by wifi connection losts) and i need to choose the color manually again (re-send DMX signal)

I've tried WLED (with DMX output compiled and pins changed), proper configured (i hope🤣) and the web ui shows  "receiving ArtNet signal" message, but DMX output not work. Tried with QLC+ artnet output and Home Assistant ArtNet integration. Both apps and both wemos D1 (Lite and pro) works whit Mtongz code, but no with WLED

What PIN and/or PIN configuration i should use? I have wemos D1 mini, wemos D1 pro, with same result..

My setup is:

DMX to Artnet ----wifi----> [Wemos + RS485 converter] ---- wired DMX out----> [RGB DMX Led strip controller] ----> RGB LED strip (analog)

Thanks in advice.

Aircoookie, 2021-08-06:
Hi!
The configuration you describe should work in principle.

Is your MAX485 connected to pin GPIO2 (D4)? And what have you set in the DMX settings menu?

djmuerteitm, 2021-08-06:
As Wiki sais, hardware part "must be" like Mtongz describes (TX pin connected to MAX).

Is DMX output assigned to D4 in WLED? I will try as you say using D4

Thanks 😊

Aircoookie, 2021-08-06:
Indeed, D4 :)

closed 2022-04-16 by Aircoookie