wled/WLED #2119: Segment brightness not saved in presets

nickion, 2021-08-06:
Brightness of segments appears not to be saved in presets. In a preset I have 4 segments for an sk6812 strip. I reduced brightness on a couple of segments, updated the preset, then ran the preset and the segments were back at full brightness. I also tried using apply on the segments before updating the preset and this made no difference. 

This is with the Toki build on a D1 mini.

Aircoookie, 2021-08-06:
Hi, thank you for reporting this!

When saving the preset after adjusting the brightness, did you tick the "Overwrite with state" checkbox?
Without this, the preset content remains the old one (this allows updating e.g. only the preset name or changing the command manually)

nickion, 2021-08-06:
Thanks for the quick update and sorry, this is my error. It works fine. I had checked it when first setting up presets and segments yesterday, but somehow completely missed it when saving adjustments.

closed 2021-08-06 by nickion