wled/WLED #2121: Additional Time Zone: AKST (Alaska Standard Time)

lefesj, 2021-08-06:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
Problem - AKST does not exist in the time zone settings for NTP config.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
I would like the AKST (UTC -9) added to the list of time zones.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
Only alternative is manual entry of mins to sec conversion 

**Additional context**
Add any other context or screenshots about the feature request here.

Thank you for your ideas for making WLED better!

Aircoookie, 2021-08-06:
Hi!
Sure, that can be added.
Just to clarify, you are looking for AKST/AKDT with daylight saving time (which would make your time zone UTC-8 currently until Nov. 7 at 2am), correct?

lefesj, 2021-08-07:
Yes, exactly!  Thank you!

closed 2021-10-14 by blazoncek