wled/WLED #2122: info.u.PIR has no existing object

Rushmed, 2021-08-06:
Hello.

I am unsing the @blazoncek PIR mod and ioBroker.
In ioBroker I get these log outputs:
`
<html>
<body>

wled.0 | 18224 | 2021-08-06 22:25:12.403 | warn | State "wled.0.e8db849b8ebe._info.u.PIR <i class="icons">&#xe325;</i>" has no existing object, this might lead to an error in future versions
-- | -- | -- | -- | --

</body>
</html>` 

I did not find an issue place on the @blazoncek Github place.

Aircoookie, 2021-08-06:
Hi! This is an extra object the PIR usermod adds to the API.
ioBroker does not know this object, hence the warning. It should be no problem to ignore it. I have no clue why ioBroker warns for extra values in the JSON. If values that were expected are missing, that would be a different story.

foxriver76, 2021-08-07:
This is an issue of https://github.com/DrozmotiX/ioBroker.wled and not of the WLED firmware

DutchmanNL, 2021-08-10:
@Aircoookie issue can be closed, fixed in adapter version v0.5.7

closed 2021-08-10 by Aircoookie

Rushmed, 2021-08-10:
Still got:
`
<html>
<body>

wled.0 | 5357 | 2021-08-11 00:12:54.892 | warn | Used invalid characters: wled.0.e8db849b8ebe._info.u.PIR <i class="icons">&#xe325;</i> changed to wled.0.e8db849b8ebe._info.u.PIR _i class=_icons_&#xe325_/i_
-- | -- | -- | -- | --

</body>
</html>`

DutchmanNL, 2021-08-11:
@Rushmed as this has nothing to do with WLED, please report your issue with my ioBroker adapter at the adapter GitHub not here :)

@Aircookie
As this a strange key, I cannot make an object of it in ioBroker : 
{"PIR <i class="icons">&#xe325;</i>" : "inactive"}

Does it have always the same value or can PIR <i class="icons">&#xe325 change ? if yes I need to remove it completely from ioBroker JSON reading and handle it different.
What does this value do ?

blazoncek, 2021-08-11:
Why are you using info object at all? 😮

DutchmanNL, 2021-08-11:
> Perhaps switching to MQTT would be better for this IMO.

definitie not, we use web socket which is much better

for now I removed this value, if not needed we can keep it like that