wled/WLED #2124: Addressing the white led after a change on an rgbw led aplication

ThatBigPrint, 2021-08-08:
**Describe the bug**
When you make a change with a RGBW led strip on the accurate mode it has a rather ungraceful step toward the end of the transition or power on 

**To Reproduce**
Steps to reproduce the behaviour, if consistently possible

Put led mode into Accurate but I think it affects dual etc I believe the current limit mode is somewhat playing a part in this but I think it should be addressing the white led chip during the transition, not at the end 

**Expected behaviour**
smooth transition without illuminating the white to only turn it off or on causing a step just at the end of a transition 

**WLED version**
 - Board: ESP32 dev
 - Version: most recent
 - Format: Binary

**Additional context**
Anything else you'd like to say about the problem?

Thank you for your help!

blazoncek, 2021-08-23:
Can you rewrite the issue in more detail?
I do not understand your description.

Aircoookie, 2022-04-16:
Please reopen if still an issue on v0.13.1! White handling has undergone a substantial update in b6 and b7.

closed 2022-04-16 by Aircoookie