wled/WLED #2126: how to mirror / reverse a segment by setSegment()?

deepminds83, 2021-08-09:
Hi,
First of all, wled is a great project and i am happy to to "play" with it.
Now i start playing around with custom usermods.
I set up 10 segments at the usersetup loop with the strip.setSegment().  This function only gets parameter like group and space. Is there a way to mirror / reverse a segment by setting it up? 
I didn't find a fungtion to set segment x as mirror or reverse.
Thanks for helping me out and sorry for ma bad and rosty english....
Deepminds

huggy-d1, 2021-08-09:
WLED v 0.12.x and 0.13.0 offer mirror/reverse ordering of segments in segment creation and setup.

Aircoookie, 2021-08-09:
Hi,
to set mirror and reverse from code, please use e.g.:

```cpp
WS2812FX::Segment& seg = strip.getSegment(segnum);

seg.setOption(SEG_OPTION_REVERSED, false);
seg.setOption(SEG_OPTION_MIRROR  , true);
```

Hope it helps :)

deepminds83, 2021-08-10:
Hi!
This is exactly what I need.
Thank you!

closed 2021-08-18 by Aircoookie