wled/WLED #2128: DMX Output not working on ESP32

windows10luke, 2021-08-09:
**Describe the bug**
When using an ESP32 with the MAX485 chip, DMX doesn't work at all.

**To Reproduce**
Compile WLED with DMX support, and adjust the DMX Output settings for the fixture you want to use. These exact same steps work perfectly on an ESP8266, but fail on ESP32.

**Expected behavior**
DMX Output is supposed to output a DMX signal

**WLED version: 0.13.0-b2
 - Board: ESP32-WROOM
 - Format: self-compiled

It appears as though same work has been done already to try to solve this issue as referenced in #817, but it was never merged.

Thank you for your help!

max9901, 2021-11-15:
A great option would be to use:
https://github.com/luksal/ESP32-DMX

good experiance with this lib, and easy to implement. If appreciated i can make a PR to implement it into Wled.

Cheerz,
Max

lplassman, 2022-05-06:
Fixed with #2652. Closing

closed 2022-05-06 by lplassman