wled/WLED #2129: Date range support for timed presets

Aircoookie, 2021-08-09:
**Is your feature request related to a problem?**
Having time controlled presets that should only apply on certain dates (e.g. holidays, appointments) requires manually activating the timed preset slot on the beginning of the date range and manually disabling it at the end of the date range.

**Describe the solution you'd like**
Allow specifying a date range (day + month, year support would be overkill) for a timer slot to be active.
For example, a Holiday season timer could turn on Colortwinkles every night at 8pm from the 1st of December to the 6th of January.

**Additional context**
This will use 3-4 bytes of RAM per timer slot. Potentially only the first few slots should support setting a date range.
This will clutter the UI. Perhaps the date + day of week controls should be collapsible.

Originally suggested by DrZzs and Xj650t on forum.wled.me

blazoncek, 2021-08-10:
I would add additional fields/dropdowns (below current timed presets) to allow selection of date and time slot. With a button to add/remove dates.
I.e.
[Jan] [1] [Slot 3] [+][-],
[Jan] [1] [Slot 4] [+]

These could be only possible if NTP is selected.

gls7, 2021-12-24:
And allow them and the other scheduled presets to be set via JSON 😄