wled/WLED #2130: Doubt about energy

pGommes, 2021-08-11:
I'm using a nodemcu on the PC, after turning on the PC the leds don't turn on automatically I have to open the application and manually turn them on, is there any way to turn on automatically whenever I receive energy?

sansillusion, 2021-08-11:
Save a preset with the color/effect/brightness you would like on startup.
Then go in config/led settings, and put the preset number in "apply preset
[xx] at boot"

>

fusionstream, 2021-09-18:
Check that power is cut to the nodemcu when you shut down your pc. I have the same issue as you because power isn't actually killed but it is sort of bypassed because I have a little script running that turns it back on once it's fully booted up.

Note that the state is actually already "on". It's just that the script calls a preset which sort of resets everything.

closed 2022-04-16 by Aircoookie