wled/WLED #2132: Option-Fields for custom-IR-codes

Rycoviac, 2021-08-11:
**Is your feature request related to a problem? Please describe.**

Like written in the wiki, I have a KY-22 laying around and a 24key rgb infrared remote. Brigtnes, on/off and all the colors works fine, bute the strobe, flash, fade and smooth buttons (function buttons) spill out an Error 12: Preset not found.
Wiki says, you can read out the custom codes with IDE, but THEN you have to change files and recompile.
I think it would be better to get rid of recompiling.

**Describe the solution you'd like**

If we can add the readout ir-codes (from the IDE) in the options menu for IR remote with custom fields like we add different led-stripes and can set specific preset-numbers for this code, it should get totaly rid of the recompiling most of the people can't or won't do.
make a + Button like led-strip-setup for another code.

**Describe alternatives you've considered**

Maybe show all possible/known ir-codes for the given remote from the dropdown menu and let user set with dropdown different options what the button should do. maybe as dropdown or just preset-numbers and -1 for default since you can adjust it all as API in presets. This is more userfriendly because you don't have to read out the IR-code. But with this option, you can't use any remote that isn't listed.

**Additional context**
-

Thank you :-)

blazoncek, 2021-08-15:
This is done with JSON IR mode.

If you want to use white 24key IR remote, just create 4 presets with ID 1, 2, 3 and 4 and Strobe, Flash,Fade and Smooth will select those presets.

Rycoviac, 2021-08-16:
Oh, thanks, I didn't know that. At least I can now set these 4 buttons. But I wouldn't close this issue jet, because an option for remapping still would be nice since I don't need all colors on my remote for example and would remap to another presets. Without compiling.

blazoncek, 2021-08-16:
You can do that with JSON IR.

Rycoviac, 2021-08-17:
Looking at https://github.com/Aircoookie/WLED/pull/1941/commits/754e3e092a4b5e2fba4a5a0090c9d42ca7546d0d
and https://github.com/Aircoookie/WLED/wiki/Infrared-Control
wiki seems outdated since I cant find the mentioned ir.json. But you lead me to the correct path, thank you :-)

blazoncek, 2021-08-17:
Indeed it is. I have asked the original author of JSON IR remote to update Wiki.
Meanwhile you can use this sample to create your own JSON and then upload it to ESP.
```
{
  "desc": "24-key Grey Presets",
  "0xFFA05F": { "label": "+", "pos": "1x1", "cmnt": "Brightness +", "cmd": "A=~8" },
  "0xFF20DF": { "label": "-", "pos": "1x2", "cmnt": "Brightness -", "cmd": "A=~-8" },
  "0xFF609F": { "label": "IX+", "pos": "1x3", "cmnt": "FX Intensity +", "cmd": "IX=~8" },
  "0xFFE01F": { "label": "On/Off", "pos": "1x4", "cmnt": "On/Off Toggle", "cmd": "T=2" },
  "0xFF906F": { "label": "SX+", "pos": "2x1", "cmnt": "FX Speed +", "cmd": "SX=~8" },
  "0xFF10EF": { "label": "SX-", "pos": "2x2", "cmnt": "FX Speed -", "cmd": "SX=~-8" },
  "0xFF50AF": { "label": "IX-", "pos": "2x3", "cmnt": "FX Intensity -", "cmd": "IX=~-8" },
  "0xFFD02F": { "label": "W", "pos": "2x4", "cmnt": "White", "cmd": "R=255&G=255&B=255&W=255&FX=0" },
  "0xFFB04F": { "label": "P01", "pos": "3x1", "cmnt": "Preset 1", "cmd": "PL=1" },
  "0xFF30CF": { "label": "P02", "pos": "3x2", "cmnt": "Preset 2", "cmd": "PL=2" },
  "0xFF708F": { "label": "P3", "pos": "3x3", "cmnt": "Preset 3", "cmd": "PL=3" },
  "0xFFF00F": { "label": "FX+", "pos": "3x4", "cmnt": "FX+", "cmd": "FX=~1" },
  "0xFFA857": { "label": "P4", "pos": "4x1", "cmnt": "Preset 4", "cmd": "PL=4" },
  "0xFF28D7": { "label": "P5", "pos": "4x2", "cmnt": "Preset 5", "cmd": "PL=5" },
  "0xFF6897": { "label": "P6", "pos": "4x3", "cmnt": "Preset 6", "cmd": "PL=6" },
  "0xFFE817": { "label": "FX-", "pos": "4x4", "cmnt": "FX-", "cmd": "FX=~-1" },
  "0xFF9867": { "label": "P7", "pos": "5x1", "cmnt": "Preset 7", "cmd": "PL=7" },
  "0xFF18E7": { "label": "P8", "pos": "5x2", "cmnt": "Preset 8", "cmd": "PL=8" },
  "0xFF58A7": { "label": "P9", "pos": "5x3", "cmnt": "Preset 9", "cmd": "PL=9" },
  "0xFFD827": { "label": "Pal+", "pos": "5x4", "cmnt": "Palette +", "cmd": "FP=~1" },
  "0xFF8877": { "label": "P10", "pos": "6x1", "cmnt": "Preset 10", "cmd": "PL=10" },
  "0xFF08F7": { "label": "R11", "pos": "6x2", "cmnt": "Preset 11", "cmd": "PL=11" },
  "0xFF48B7": { "label": "P12", "pos": "6x3", "cmnt": "Preset 12", "cmd": "PL=12" },
  "0xFFC837": { "label": "Pal-", "pos": "6x4", "cmnt": "Palette -", "cmd": "FP=~-1" }
}
```
The first key is Hex code of a button (needs to be in CAPITAL). The important second key is "cmd" which contains actual API command (JSON or HTTP). "label", "pos" and "cmnt" are just comments to help you mentally map buttons to keys.

Aircoookie, 2021-08-17:
Info by @scottrbailey now merged to https://kno.wled.ge/interfaces/json-ir/json_infrared/ :)

closed 2021-08-17 by Aircoookie