wled/WLED #2133: NodeMCU/WS2812B Not working with WLED

ahmaad-ansari, 2021-08-11:
I am currently setting up a LED project and ran into some barriers. One moment the LEDS were running perfectly and the next they started glitching and then turned off. I am using an ESP8266 with a 5V 40A Power Supply. The LED strip does not turn on but I can use the WLED interface from my phone and laptop. I checked if power is being supplied to the strip, which it is, so I think the problem lies within data transfer. This is my first time doing such a project so any help would be appreciated.

Rapti, 2021-08-13:
First, if you haven't already, make sure the data cable is correctly attached as well. What you are describing could be caused by a loose connection.

Another possibility is that the data signal does not have a high enough voltage. The ESP8266 runs at 3.3 V while the LEDs run at 5 V. This means that the data signal is technically to low to be recognized as HIGH by the first LED's built-in chip. This has never been a problem for me personally, but it could still be what you are experiencing, especially if your power supply provides slightly more than 5 V. See [My LEDs act funny and flicker randomly - Reason 2](https://github.com/Aircoookie/WLED/wiki/FAQ#my-leds-act-funny-and-flicker-randomly) in the FAQ. If that really is your problem, I recommend checking out the [workaround link](https://hackaday.com/2017/01/20/cheating-at-5v-ws2812-control-to-use-a-3-3v-data-line/) at the bottom. You'll only need a single extra diode or resistor for that.

udummy, 2021-08-30:
I found out that the PIN numbers don't line up.  Pin2 in settings is pin 4 on nodemcu.

closed 2022-02-06 by ahmaad-ansari