wled/WLED #2136: Sync Groups

smartroad, 2021-08-14:
When clicking the Sync button it syncs all the current WLED installs to the current unit. Is there a way to set groups for syncing? I have several WLED lights in various rooms and would like to have them as seperate groups to sync to.

I was thinking in the settings there could be an option for setting a sync group number just 0-9 (although maybe 0-255 as it would have to be a byte anyway?).

Dunno if that sounds like a good idea or not, or if I have somehow missed an existing feature!

itCarl, 2021-08-15:
You could simply change the upd port of the corresponding wleds you want to have in a group.
Settings > Sync Interfaces/Setup > UDP Port

smartroad, 2021-08-15:
> You could simply change the upd port of the corresponding wleds you want to have in a group.
> Settings > Sync Interfaces/Setup > UDP Port

Would that affect any other functionality of the system? Using external programs (i.e. LEDFX) or identification of other instances from the web interface. 

Actually just made me think, given WLED can find other instances on the network, when clicking on the sync maybe it could ask which other instances to attach to. Would be a more user friendly focused implementation compared to changing group or port numbers.

Aircoookie, 2021-08-16:
Changing the UDP port does not impact other functionality :)
The secondary UDP port also runs the instance discovery as well as being the default port for the tpm2.net protocol.
The primary port can be changed freely. I agree that a smaller group number would be more intuitive, that might be a good idea.

Directly creating sync groups on the device would be awesome. A challenge would be that the device would need to remember the instances to sync to even if they become unreachable for some time and this would likely cost a not insignificant amount of memory. Nevertheless, adding a concept of "Rooms" would definitely make using multiple WLED devices together a lot more accessible.

smartroad, 2021-08-17:
Given the current sync method, where a "master" node initiates the sync, could the client nodes just remeber the Masters ID and then ping it when back online? The Master could reply with:

"Sure still grouped here are your settings"
or
"Not in my group anymore do your own thing"

I am a hobbyist coder so I am probably way over simplifying it lol but would a simple protocol like that work?

pbolduc, 2021-08-20:
I think ask @itCarl said, if you want group 2 or more instances to be sync'ed, you can use the same UDP port across those instances. This is definitely less user friendly than being able to define groups. If there were one or more free bytes in the UDP packet that is used in the sync message, it could use bit flags to indicate the target rooms/areas, ie 00000001 => only those instances registered for room one would apply this message. With a single byte, you could support 8 rooms/areas with any combination. Across your home, you would need to ensure your room indexes are consistent. It would be more expensive to try to support sending room names across the sync packet. 

UPD sync message formats:
```
version = 0 - sync primary color, nightlight active/minutes, brightness (if enabled)
version > 0 - sync white value
version > 1 - sync secondary color
version > 2 and < 200 - sync effect intensity
version > 3 - sync transition delay
version > 4 and < 200 - sync effect palette
version > 5 and < 200 - sync strip time base
version > 6 - sync tertiary color
version > 7 - sync time
version < 200 - sync current effect and effect speed
```

blazoncek, 2021-08-21:
@pbolduc this is actually a brilliant idea.
If we extend (or otherwise squeeze out a byte or two form) UDP packet we can just add two fields/checkbox groups on the Sync settings page one for "Send to groups" and another one for "Receive from groups". And then we add the first bit-field to the UDP packet and use the second within the receiving logic using bit-wise AND to check if the message is intended for the receiver.
This wouldn't add any significant overhead neither to sending nor receiving logic and would only use a byte or two of permanent RAM.

blazoncek, 2021-08-23:
Sync groups are tested and working as expected (see #2150 ). They will be available in next beta release.
Older devices will disregards sync groups and will continue to listen on all groups. You will need to update all of your devices to have full control of groups in your environment.
There is maximum of 8 available groups with group 1 being default on new installs or upgrades.

blazoncek, 2021-08-30:
This is now part of master and will be available when 0.13 is released.

closed 2021-08-30 by blazoncek