wled/WLED #2137: One strip flickering on WLED 0.13.0-b2

pszafer, 2021-08-15:
**Describe the bug**
Hi, I have 2 LED strips connected with Wemos D1 Mini on ESP32. Both strips have 270 LED's.
Before I used WLED0.11 with esp32_multistrip mod and most of the time it was working great.
From time to time it hanged and reset.

I wanted to upgrade to the newest WLED to test usage of multiple LED strips without mod.
Unfortunately it is not working very well.

My setup is as below:
[image]
Problem is always with Segment 1.
Even if I change config order than problem is with Segment 0.

- small problem: From time to time transition occurs only on one strip (Segment0).
- second problem is that strip is flickering. Like I want to change color to blue, but in first miliseconds I see blink with different color and than choosen color.
- after a while (half an hour) I see that Segment1 is flickering (I see brighter color).

I couldn't reproduce problem with effects enabled.
Reverting to WLED 0.11 with usermod and problem is gone.

**To Reproduce**
Choose one color on both segments.

**Expected behavior**
A clear and concise description of what you expected to happen.

**WLED version**
 - Board:Wemos D1 ESP32 / Quinled Dig Uno
 - Version 0.13.0-b2
 - Format  binary

**Additional context**
Anything else you'd like to say about the problem?

Thank you for your help!

blazoncek, 2021-08-17:
Flickering is a known bug with 0.13-b2.
You can try builds made by @srg74 which implemented a workaround.

sansillusion, 2021-08-17:
I figured it was a recent update that did that. My two wled never had
flickering issues before. Will be waiting for a fix.

pszafer, 2021-08-17:
thanks for feedback, I thought that somehow I did something wrong

blazoncek, 2021-08-19:
@sansillusion @pszafer can you test the build I mentioned? Please.
We are still testing if the workaround does work for all users.

pszafer, 2021-08-19:
sure, this fork? https://github.com/srg74/WLED , master branch?

blazoncek, 2021-08-19:
> 
> sure, this fork? https://github.com/srg74/WLED , master branch?

He has a separate folder for firmware releases.

srg74, 2021-08-19:
Firmware bins https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware/WLED_wemos_shield/v0.13.0-b2

pszafer, 2021-08-22:
I'm testing it and it's working very well. Maybe there is some flickering but I'm not sure (for 2 days).
One question though, is it possible to have both segments in HA as single light similar as it was with multistrip mod (master light in HA doesn't support colors and effects)?

FigJam23, 2021-08-23:
Hey guys this my first WLED setup using RGBW lefs with a esp32 5v and capacitor unfortunately I don't have a level shifter yet as you can see in my video below I'm getting lots of flickering and color changes while set to low brightness white. The flickering happens across all colors and modes. Would this be because of the level shifter not being used or potentially a Fw issue like above post might suggest. 
If I do try a different version .bin what should I try and is there a way to backup my settings and restore too new fw .
Thanks in advanced 

https://photos.app.goo.gl/gtNWr8DhYRtnnNyT6

blazoncek, 2021-08-24:
@FigJam23 flickering is usually caused by incompatible voltage levels between ESP and LED strip or bad wiring. Use level shifters (a good one is 74AHCT125) and check your connections and wires.

There is a known issue in 0.13-b2 pre-compiled binary available but has been since identified and (most likely) resolved. Fix will be available in next pre-compiled beta binary update (0.13-b3)
@pszafer @sansillusion if you still see flickering with latest (or @srg74) builds please feel free to reopen the issue.

closed 2021-08-24 by blazoncek