wled/WLED #2138: Estoy buscando mi dispositivo en wled, pero aparece este mensaje "found not lights yet"

YaenM, 2021-08-16:
Hola Comunidad, la verdad soy muy nuevo en este tema, el problema es el siguiente, ya configure mi Wled con un repetidor que tengo cerca de mi cuarto que está a 2.4ghz, al principio me reconoce el dispositivo pero después de que salgo de la aplicación de wled de mi celular, ya no responde, entonces borro el dispositivo que entrontro para volverlo a generar pero ya no me aparece y me aparece el mensaje de "found not lights yet" y lo único que me funciona es aplastar el botón de resetear que incorpora el esp8266, 
[image]

blazoncek, 2021-08-17:
English, please.

spike62, 2021-08-17:
192.168.4.1 is default IP, Looks like you have not setup your Wifi Access Point and attached to a WIFI network.

Try attaching to a WifI netork  then browse for lights.

04aromeo, 2021-08-29:
Hola buenas, a mi tampoco me aparecía, tuve que configurar su propia IP y ponerla de manera manual, para cualquier cosa, escribeme

closed 2022-04-16 by Aircoookie