wled/WLED #2139: WebSocket: Implement "__Pong__" response or "alive" message at interval

DutchmanNL, 2021-08-16:
**Is your feature request related to a problem? Please describe.**
Thanks to WLED 0.10.x we can use WebSocket to receive live data immediately, unfortunately only data is send if a value changes and the WebSocket will not ache a connection failure or time-out.

As a workaround I query the HTTP-API to see if I get an response, this is basically unneeded network traffic and generation of the JSON.
Can you please implement and "__Pong__" answer to the WS as a reply so we can build a watch dock fully in WS ?
Or provide in regular intervals status like "Alive" so my code is aware f connection failures.

This is also used as best-practice for websocket connections called the Ping/Pong methods

**Describe the solution you'd like**
Fully communication by websocket instead of HTTP

**Describe alternatives you've considered**
I can handle a request by http, but its not that efficient and unneeded generation oof JSON by the ESP

Aircoookie, 2021-08-16:
Hey!

I've just checked and a pong reply to a ping packet should be sent :)
https://github.com/Aircoookie/ESPAsyncWebServer/blob/master/src/AsyncWebSocket.cpp#L691

If that does not work, websocket clients/libraries normally fire an event on disconnect or error, which you could use.
You can also send `{"v":true}`, that will prompt an immediate response (if the connection is working) and at least you won't have to use HTTP to verify.

Lastly, I could implement such an alive message (perhaps send a ping from my end?), but that is, in most cases, unneeded traffic as well, so I'd like to avoid it if possible.

DutchmanNL, 2021-08-16:
> I've just checked and a pong reply to a ping packet should be sent :)
> https://github.com/Aircoookie/ESPAsyncWebServer/blob/master/src/AsyncWebSocket.cpp#L691

Hmm interesting, I don't get a message back by sending a ws.ping :/
Sending it every 5 seconds, no message response (state updates are received !

> If that does not work, websocket clients/libraries normally fire an event on disconnect or error, which you could use.

that was my though to, but if you have an active WS connection and (for example) cut the power of WLED ESP I don't get an error or socket closed event

> Lastly, I could implement such an alive message (perhaps send a ping from my end?), but that is, in most cases, unneeded traffic as well, so I'd like to avoid it if possible.

fully agree, ping/pong would be best practice in my eyes

blazoncek, 2021-08-17:
I have noticed that sometimes WS stays open but no data is transmitted (only in one way).
I have implemented WS close and reconnect in WLED UI for such cases.

DutchmanNL, 2021-08-17:
> I have noticed that sometimes WS stays open but no data is transmitted (only in one way).
> I have implemented WS close and reconnect in WLED UI for such cases.

interesting, bit not workable for me as the UI is never used all controlled by the SmartHome system which should be aware of a WS disconnect :)

Aircoookie, 2021-08-17:
Alright, I have checked and it seems like ping/pong is commonly a low-level functionality not exposed to most APIs (e.g. the browser javascript one). I couldn't even find a WS client that let's me test ping/pong.

I have decided to replicate the functionality on the application layer, you can now just send the string `ping` instead of a JSON object and WLED should respond with `pong`.

closed 2021-08-23 by Aircoookie

DutchmanNL, 2021-08-25:
[image]

confirmed, works perfectly thank you!   now I. can finalize that stuff