wled/WLED #2141: Remote API call timeout

0rsa, 2021-08-18:
Wled 0.12.0

I use a simple PHP script which is calling WLED Api using curl (I have 4 different lolin mini d1 pro, all are impacted by this issue)

`$wledData = array('seg' => $state->seg);

$ch = curl_init( "http://".$light->ip."/json/si" );
# Setup request to send json via POST.
$payload = json_encode($wledData);
curl_setopt( $ch, CURLOPT_POSTFIELDS, $payload );
curl_setopt( $ch, CURLOPT_HTTPHEADER, array('Content-Type:application/json'));
# Return response instead of printing.
curl_setopt( $ch, CURLOPT_RETURNTRANSFER, true );
# Send request.
$result = curl_exec($ch);
curl_close($ch);`

Sometimes it works perfectly, sometimes it starts to timeout and it is the beginning of a timeout loop.
If I connect to web interface, commands are working fine without delay.

Same issue has been reported here:
https://github.com/danecreekphotography/node-red-contrib-wled2/issues/70

Thank you

pbolduc, 2021-08-20:
Are you running on ESP32 or ESP8266?

0rsa, 2021-08-20:
I use ESP8266

Aircoookie, 2021-08-20:
Hey!
I am not familiar with PHP, but what is the JSON payload you are trying to send? Seems to be the segment array, but how are you generating that?
How often are you sending the request to WLED?
You can try enabling the "Disable WiFi sleep" option in WiFi settings, sometimes this helps improve reliability.

0rsa, 2021-08-20:
Yes, I send only segment array which is generated by more PHP code but I assume structure is correct, it works most of the time.
About Wifi Sleep, it worth a try but I'm skeptical because I can access to the Web UI and send commands when my PHP script is at the same time in timeout.
Thank you

blazoncek, 2021-08-20:
How many simultaneous request do you make?
Anything more than 1 is risky on 8266.

0rsa, 2021-08-20:
It's difficult to estimate.
I was calling WLED states every minutes all the time to get fresh state (so better using MQTT with events).
All my lights are automated (including WLed), scenes, colors, brightness are automatically adjusted in a day (brightness is calculated with outdoor brightness sensors, example: 6000 lumens => 20% brightness).
I can also control all my WLed with my own interfaces (Desktop + Mobile).

So I have no doubt that sometimes, several calls arrive at the same time.

2 questions:
- Is MQTT more robust to handle a lot of queries?
- Is a ESP32 more robust to handle a lot of queries?

Thank you guys

blazoncek, 2021-08-20:
What you are describing calls for websockets implementation on your part. Just subscribe to websockets (ws://wled_ip/ws) and you'll receive JSON updates. No need for polling.

0rsa, 2021-08-20:
I missed this feature, thank you very much.
I'll move to websockets implementation asap.
Thank you again (I close this thread considering issue is resolved as "don't make simultaneous calls")

closed 2021-08-20 by 0rsa

0rsa, 2021-08-21:
Done.
Hoping ws will be reliable :)
Thanks!

[image]

GSVNoFixedabode, 2022-04-01:
@0rsa, any chance of sharing the full example of the php code you used to POST the JSON to WLED? I'm trying to get an infrequent php call but getting nowhere fast.  Thanks.

0rsa, 2022-04-01:
@GSVNoFixedabode I was using this portion of code

`$ch = curl_init();
        curl_setopt($ch, CURLOPT_URL, "http://WLED_IP/json");
        curl_setopt($ch, CURLOPT_RETURNTRANSFER, 1);
        $output = curl_exec($ch);
        curl_close($ch);`