wled/WLED #2142: WLED not responding to MQTT

Thotho131097, 2021-08-18:
Hi all,

I am a newbie to WLED and MQTT. I've tried to setup my MQTT towards WLED, yet I am unable to get response.
I've used the same settings for my SONOFF MINI + RF BRIDGE (except for different topics etc) and this is working fine.

Can anyone see a mistake I am making? 
WLED settings:
[image]

Home assistant / NODE red settings:
[image]

Message. payload settings:
[image]

Testsetup is basicly a inject node, message gets transformed to "Toggle" and sent over MQTT towards WLED:
[image]
Yet WLED does not seem to receive the message :( 
What am I missing here?

victorclaessen, 2021-08-18:
Can you test with another tool (like MQTT Explorer, or mosquitto_sub) to see if the messages you expect WLED to receive are really being published on the broker?

blazoncek, 2021-08-19:
do not use same topic for device and group.

Aircoookie, 2021-08-19:
Dumb question, but have you rebooted WLED after adding your MQTT configuration?

closed 2021-08-19 by Thotho131097