wled/WLED #2143: Hue status: Invalid light ID

luisminst, 2021-08-18:
Hello 

I am trying to sync led strip WS2811 with my hue bridge but alwais I receive the same error: Hue status: Invalid light ID

In other forums I have read that it is necessary to check the ID in the philips hue app but I cannot find this ID and I do not know what to write in "Poll Hue light"

Can anybody help me please? my firmware version is WLED_0.11.1_ESP8266_ledpin4.bin

thanks in advance

[image]

Aircoookie, 2021-08-18:
Hi!
My apologies, it appears as if the Hue app removed the Light ID info in a recent update.
If you have access to an Android device, you can use the app "Hue config viewer" and go into the lights menu to see a list of your hue lights and their ID numbers.
I will need to add a similar functionality to WLED in the medium term.

luisminst, 2021-08-19:
Thanks for your quick response. I have installed the app "Hue config viewer" but I cant found my new led strip WS2811 (conected to WLED controler). I can only see the original philips lamps in the app.
In the Hue application I can't find the led strip either. What else could I try?
Thanks a lot

Aircoookie, 2021-08-19:
That is unfortunately not how the hue sync functionality works, the WLED led strip can't be controlled from the hue app individually.

Rather, WLED syncs _to_ a hue light you determine, meaning it will always have the same color/brightness. In hue config viewer, there is a number to the left of each of the lamps. Choose the one you want WLED to sync to and enter its number in the WLED setting "Poll hue light".

closed 2022-04-16 by Aircoookie