wled/WLED #2145: my /edit page is not accessible

neckyaan, 2021-08-19:
Version 13.0-b2 http://[controller-ip]/edit page is not accessible，my Lock wireless (OTA) software update is open。 [controller-ip] is 192.168.*.*

Aircoookie, 2021-08-19:
Hi!
Sorry about that. What error does the browser give, a timeout or 404?
It is also possible that it does not load if you don't have internet access on your device (e.g. when connected to WLED-AP), as it needs to load the `ace` library: https://github.com/Aircoookie/WLED/issues/1592

neckyaan, 2021-08-19:
[image]
[image]

neckyaan, 2021-08-19:
the browser give me nothing pormpt

Aircoookie, 2021-08-19:
Seems like it is indeed having problems loading the ace library. We will need to modify the `/edit` page to fallback to a simple textarea if ace cannot be loaded. I believe @embedded-creations recently had a look at a fork of ESPAsyncWebserver which improves the `/edit` page.

blazoncek, 2021-08-19:
please show your Network tab after reloading /edit page

embedded-creations, 2021-08-19:
> I believe @embedded-creations recently had a look at a fork of ESPAsyncWebserver which improves the `/edit` page.

This issue is still present

neckyaan, 2021-08-20:
Sorry for the late response
I found the problem. There are network restrictions in my area, which prevents me from accessing https://cdnjs.cloudflare.com/ajax/libs/ace/1.2.6/ace.js

Aircoookie, 2021-08-23:
Alright :) I'll close this as it is just another manifestation of #1592 , but that doesn't change the fact that we should add a fallback if the library can't be fetched.

closed 2021-08-23 by Aircoookie