wled/WLED #2146: MQTT: XML data missing power state information

0rsa, 2021-08-20:
Hi,
Please add an attribute to know if WLED is ON or OFF in XML data
A single attribute containing brightness value is not enough, you can have a WLED OFF with brightness set to a value > 0, we miss one information with a sinle attribute for toggle/brightness.
Thank you

Refering to: https://github.com/Aircoookie/WLED/wiki/HTTP-request-API section XML response

blazoncek, 2021-08-20:
It is not possible to have bri(ghtness)>0 and have WLED turned off or the other way around (having bri==0 and WLED on).

If global brightness equals 0 it means WLED state is OFF. If it is not equal to 0 then WLED state is ON.

0rsa, 2021-08-20:
@blazoncek I have this JSON history in my database logs

`{"state":{"on":false,"bri":29`

Master Off/On/Toggle is "OFF" (false)
Master brightness is 29

If brightness and toggle are the same, how retrieve your previous brightness when you toggle on your WLED?

blazoncek, 2021-08-20:
It is internally stored and retrieved when needed.

What you are seeing in JSON is stored value.
```
root["on"] = (bri > 0);
root["bri"] = briLast;
```

0rsa, 2021-08-20:
Ok, thank you.
Why not to share this internal value in XML data?
Why XML works in a different approach of JSON?

blazoncek, 2021-08-20:
It doesn't. JSON just gives you additional information.
Some things are difficult to implement in JSON some in XML/HTML. You can freely mix the two in your API calls.

0rsa, 2021-08-20:
Sure but I'm currently switching to MQTT due to HTTP JSON issues :)
https://github.com/Aircoookie/WLED/issues/2141
Mixing requests it not an option for my case.

blazoncek, 2021-08-20:
Check if you are flooding WLED with too many requests at once.

0rsa, 2021-08-20:
Moving to WebSockets to use JSON instead of MQTT / XML.
Thank you

closed 2021-08-20 by 0rsa