wled/WLED #2148: Unable to connect to a Wifi network when WPA password contains exclamation mark "!"

GendronAC, 2021-08-21:
**Describe the bug**
I am unable to connect the WLED to my network. Password is in the form <8 letters><1 digit><1 special char> (e.g. AlloAllo3!)

**To Reproduce**
1. Factory boot on a ESP8266 2M running version 1.12.0
2. Setup wifi to my home network with said password (e.g. AlloAllo3!)

**Expected behavior**
Able to connect to the network and rock the neighborhood !

**WLED version**
 - Board: ESP8266
 - Version 0.12.0
 - Format : official release

**Additional context**
Created a secondary SSID on my network with a password without a "!" and was able to connect, so I don't think any other element is related.

Congratulation on this awesome project !

jdanbru, 2021-08-21:
I have an “!” In my WiFi password.

what’s your network setup?

GendronAC, 2021-08-21:
Ubiquity (multiple devices)

blazoncek, 2021-08-21:
Try updating firmware on your Ubiquiti APs.

GendronAC, 2021-08-21:
Using the latest. Close it if you are unable to reproduce it.

Aircoookie, 2021-08-21:
I've had multiple reports of SSIDs with special characters not working in a past.
Since "change your password" would be bad advice and a considerable effort, we need to find out what causes this and what characters are affected.

I will ask on discord if others have SSIDs with an `!` that can connect successfully.

GendronAC, 2021-08-23:
The ssid is using plain character (MaisonOiseau)

"LedDriver" is working
"MaisonOiseau" isn't

GendronAC, 2021-08-23:
[image]
[image]

Eddiiie, 2021-12-10:
I have Ubiquity, my wifi password has !  at beginning and end.   Not problems.

closed 2022-04-27 by stale[bot]